astro-pi/python-sense-hat #1: Should get_compass/accel/gyro restore IMU settings after making a call

martinohanlon, 2015-05-13:
get_compass, get_accelerometer and get_gyroscope all change the IMU config e.g.:

self.set_imu_config(False, True, False)

When the function returns should it set the config back to what it was? Otherwise subsequent calls to get_orientation will only use one of the sensors?

davidhoness, 2015-05-21:
I appreciate the argument for this however, in most cases, the consuming code will have a tight loop where only `get_compass`, `get_accelerometer` or `get_gyroscope` is being called. If we implemented this the config would be changing twice per iteration of their loop. See [this note](https://github.com/astro-pi/astro-pi-hat/blob/master/astro_pi/astro_pi.py#L526) in the main library code.

martinohanlon, 2015-05-21:
Got ya...

closed 2015-05-21 by martinohanlon