astro-pi/python-sense-hat #2: ap.show_letter('1')

babybigfox, 2015-05-31:
The function show_letter is not compatible in displaying the number 1 on the screen. Line 396 states that len(s)  > 1 when len(s) > 0 to make it compatible with all numbers. I found this issue at a Raspberrry Jam with a fellow GitHub user. All you need to do is change 1 to 0 to make it work.
Daniel Harrin age 12
babybigfox

babybigfox, 2015-05-31:
Sorry - I got that wrong - I don't know how to fix it... Ignore what I said

babybigfox, 2015-05-31:
The ap.show_letter error is still there when I display the number one. Any help?

davidhoness, 2015-06-02:
This is not an issue. I am guessing babybigfox is not in possession of an Astro Pi HAT otherwise they would have just run `ap.show_letter('1')` and seen that it does indeed work.

Line 396 of the module is testing the string *length* (using `len`) that has been passed into the function, not the content of the string. It will only throw the ValueError exception, for example, if you run `ap.show_letter('yo')` where you're sending in a string of length two.

closed 2015-06-02 by davidhoness

babybigfox, 2015-06-03:
I have an Astro Pi hat and I found the issue with an organiser of the Jam when I tried to program the pi to count down from 9 to 0 which got stuck on 2 every time. Would you be able to suggest some coding to do the count down as I am new to Python and obviously need help!

davidhoness, 2015-06-03:
Please post your code [here](https://www.raspberrypi.org/forums/viewforum.php?f=104) and I'll take a look.

babybigfox, 2015-06-04:
I have posted it...but it needs to be approved by a moderator before you can see it...

davidhoness, 2015-06-04:
Turns out this was a bug! Well done for finding it =)

So in your code you used `ap.show_message('1')` as well as `ap.show_letter('1')`. The show message function has [code](https://github.com/astro-pi/astro-pi-hat/blob/master/astro_pi/astro_pi.py#L368) in it to trim the white space from the back and front of the 5x7 text character so that long messages don't have huge gaps in them.

The internal `_trim_whitespace` function was incorrectly modifying the global copy of all the characters stored in `_text_dict`. So I've done a fix which means that the internal `_get_char_pixels` function that safeguards `_text_dict` always returns cloned copies of the 5x7 character. Then they can be trimmed by `_trim_whitespace` and discarded without it changing the original in `_text_dict`.

babybigfox, 2015-06-04:
Thanks for that update but the pi now shows an error, all to do with show_letter, that pixel lists 'must have 64 elements'. Any help?

babybigfox, 2015-06-04:
Thanks for the advice. I have don the updates and restarted my pi but it still shows the error
Traceback (most recent call last):
                                    File "ReactionPi.py", line 182, in <module>
                                                                                   menunum() # Start the whole program!
                                                                                                                         File "ReactionPi.py", line 84, in menunum
    write()                                 # Enter pressed
                                                             File "ReactionPi.py", line 159, in write
                                                                                                         ap.show_letter('1') #
                                                                                                                                File "/usr/local/lib/python2.7/dist-packages/Astro_Pi_HAT-0.0.2-py2.7.egg/astro_pi/astro_pi.py", line 378, in show_letter
                                                                                           self.set_pixels(coloured_pixels)
                                                                                                                             File "/usr/local/lib/python2.7/dist-packages/Astro_Pi_HAT-0.0.2-py2.7.egg/astro_pi/astro_pi.py", line 197, in set_pixels
                                                                                       raise ValueError('Pixel lists must have 64 elements')
                                                                                                                                            ValueError: Pixel lists must have 64 elements

when it gets to the number 2 in show_letter 1 during the countdown in my function named write()
Anything wrong with my coding?

davidhoness, 2015-06-04:
Please use the raspberry pi forums for technical support.

babybigfox, 2015-06-04:
ok