astro-pi/python-sense-hat #6: Handle i2c-dev not being loaded

XECDesign, 2015-09-02:
If i2c-dev isn't loaded, the library tells you to run as root, which doesn't help in this situation. Maybe a check for the appropriate dev node and an appropriate error message would avoid some confusion.

The sense-hat package adds i2c-dev to /etc/modules, but it's possible to just install python(3)-sense-hat without installing the sense-hat package. Or a user may end up removing that line accidentally.

bennuttall, 2015-10-02:
Is this fixed? Just closing off issues

XECDesign, 2015-10-02:
Doesn't look like it. It will still try to init the sensors and if there's any sort of error, you get a "please run as root / use sudo" message. I have seen people who have device tree disabled get this error message and get confused because 'sudo' is not the solution. I'm okay with this being closed if it isn't really seen as an issue, but thought it was worth bringing up.

viki2, 2015-12-25:
:+1: for this. I'm stuck with disabled I2c for almost 1 hour.

5chdn, 2015-12-27:
:+1: same here. see #25

closed 2016-06-06 by bennuttall

bennuttall, 2016-06-06:
Hi @XECDesign we've applied a fix but not sure it's valid. We check for presence of `/dev/i2c*` and fail on init if it's not there. How do you get to the point where it fails because I2C is disabled, so we can reproduce and check?

XECDesign, 2016-06-06:
If you comment out i2c-dev from /etc/modules and reboot, you would've hit the issue. `sudo modprobe -r i2c-dev` should also do it.

commit 8ab3a290 (2016-06-06): Handle i2c-dev not being loaded, close #6