astro-pi/python-sense-hat #7: Apt package issue

gbaman, 2015-09-02:
Just a heads up, there is an issue with the apt sense-hat package where if you don't have a /boot/config.txt file, it will fail to install.
The Pi doesn't actually need a config.txt to boot so the install script doesn't need to assume there is one.
A simple
```touch /boot/config.txt```
Before ```sed /boot/config.txt -i -r -e "s/^((device_tree_param|dtparam)=([^,]*,)*i2c(_arm)?)(=[^,]*)?/\1=on/"``` in ```postinst``` would sort it.
Was a problem with https://github.com/PiNet/PiNet as the Raspbian chroot doesn't actually have the config file given the boot files are stored in a different location than the root OS (given root OS is on a server). It does though have a dummy /boot folder.

XECDesign, 2015-09-02:
Thanks for spotting that one. Will get it fixed for the jessie release.

bennuttall, 2015-10-01:
@XECDesign did you fix this?

XECDesign, 2015-10-02:
It has been fixed in jessie, yes.

closed 2015-10-02 by bennuttall