astro-pi/python-sense-hat #8: Getting error message

Forsinogul, 2015-09-19:
I am getting the following error message: OSError: Cannot detect RPi-Sense FB device.  However all LED lights are on.

XECDesign, 2015-09-19:
Hello, could you please provide the output of the following:
```
sudo apt-get install device-tree-compiler
uname -a
vcgencmd version
sudo dtc -I fs /proc/device-tree
```

Forsinogul, 2015-09-20:
thank you for your help.  below is the output requested:
uname -a
Linux pi 4.1.6-v7+ #810 SMP PREEMPT Tue Aug 18 15:32:12 BST 2015 armv7l GNU/Linux
vcgencmd version
Aug 20 2015 17:44:52 
Copyright (c) 2012 Broadcom
version 3f87086448257aa66881de858a0b4b0e6daa8719 (clean) (release)
sudo dtc -I fs /proc/device-tree
FATAL ERROR: Couldn't opendir() "/proc/device-tree": No such file or directory

XECDesign, 2015-09-20:
You do not appear to have device tree enabled, which prevents the necessary modules from being loaded automatically. You could modprobe rpisense-fb and rpisense-js or stick them in /etc/modules.

Forsinogul, 2015-09-20:
I have enabled device tree in raspi-config and it works.  thanks

XECDesign, 2015-09-20:
Glad to hear it. Have fun.

closed 2015-09-20 by XECDesign

ARKopp, 2015-10-29:
I also have the same problem. But I am using kano

XECDesign, 2015-10-29:
I'm sorry, I only support the official foundation Raspbian distribution. I don't own or use kano and am not familiar with how they set up the software, so I won't be able to help.

bennuttall, 2015-10-30:
I saw a tweet recently that they were thinking of adding it soon. For now you can use [Raspbian](https://www.raspberrypi.org/downloads/).

ARKopp, 2015-10-30:
yes i just got it. Alejandro just send me Kano Jesse. Hope it works.