astro-pi/python-sense-hat #9: Example pygame_joystick.py does not  run headless, use python_evdev instead of pygame

tvoverbeek, 2015-09-23:
The pygame_joystick example needs a display otherwise the event loop does not work.
This is a problem when running headless (e.g. RPi with Sense Hat accessed via ssh).
Also the display is initialized with size 640x480 which is not nice to the new Raspberry Pi Display (800x400).

You can get it to work headless with 'hdmi_force_hotplug=1'  in /boot/config.txt.
However the ESC key will not be recognized via ssh. You have to terminate with Ctrl-C

The code only needs keyboard events, so a display should not be needed.
I wrote an evdev_joystick.py using the same logic as pygame_joystick.
If you are interested I can generate a pull request with my added evdev_joystick.py.
Of course, instructions will have to be added how to install python_evdev.

XECDesign, 2015-09-23:
I don't believe this is an issue, it's just a limitation of using pygame for this. As you say, using evdev gets around this but has its own limitations.

rhellyer, 2015-09-29:
I have a similar issue where I want to use my sense hat headless but allow the user to control the led screen from the joystick. I tried with pushed as in the examples but it failed because of no display. Could you post your even example somewhere or could someone recommend am alternative approach?  Many thanks

tvoverbeek, 2015-09-29:
rhellyer, easiest is to use `hdmi_force_hotplug=1` in /boot/config.txt.
This let the Pi think there is a hdmi display attached.
As in the pygame-joystick example you have to initialize the display, but a size of 1x1 is sufficient:
```
pygame.init()
pygame.display.set_mode((1, 1))
```

rhellyer, 2015-09-29:
Many thanks - that sounds like the simplest route!

davidhoness, 2015-10-01:
@tvoverbeek is it acceptable to add another version of the program for evdev? That way users can see the advantages and disadvantages of both ways.

tvoverbeek, 2015-10-03:
For those interested here is my evdev version of the pygame-joystick example.
The program logic is the same as in pygame_joystick.py.
Since evdev looks at input sources on the Pi itself, you cannot use ESC to terminate the program over ssh.
Therefore Ctrl-C instead.
```
#!/usr/bin/python
import sys
import time
from sense_hat import SenseHat
from evdev import InputDevice, list_devices, ecodes

print("Press Ctrl-C to quit")
time.sleep(1)

sense = SenseHat()
sense.clear()  # Blank the LED matrix

found = False;
devices = [InputDevice(fn) for fn in list_devices()]
for dev in devices:
  if dev.name == 'Raspberry Pi Sense HAT Joystick':
    found = True;
    break
if not(found):
  print('Raspberry Pi Sense HAT Joystick not found. Aborting ...')
  exit()

# 0, 0 = Top left
# 7, 7 = Bottom right
UP_PIXELS = [[3, 0], [4, 0]]
DOWN_PIXELS = [[3, 7], [4, 7]]
LEFT_PIXELS = [[0, 3], [0, 4]]
RIGHT_PIXELS = [[7, 3], [7, 4]]
CENTRE_PIXELS = [[3, 3], [4, 3], [3, 4], [4, 4]]

def set_pixels(pixels, col):
    for p in pixels:
        sense.set_pixel(p[0], p[1], col[0], col[1], col[2])

def handle_code(code, colour):
    if code == ecodes.KEY_DOWN:
        set_pixels(DOWN_PIXELS, colour)
    elif code == ecodes.KEY_UP:
        set_pixels(UP_PIXELS, colour)
    elif code == ecodes.KEY_LEFT:
        set_pixels(LEFT_PIXELS, colour)
    elif code == ecodes.KEY_RIGHT:
        set_pixels(RIGHT_PIXELS, colour)
    elif code == ecodes.KEY_ENTER:
        set_pixels(CENTRE_PIXELS, colour)

BLACK = [0, 0, 0]
WHITE = [255, 255, 255]

try:
    for event in dev.read_loop():
        if event.type == ecodes.EV_KEY:
            if event.value == 1:  # key down
                handle_code(event.code, WHITE)
            if event.value == 0:  # key up
                handle_code(event.code, BLACK)
except KeyboardInterrupt:
    sys.exit()
```

rhellyer, 2015-10-05:
This is just what I needed for my project as I want to use the minijoystick but don't need pygame, many thanks for posting!

closed 2015-10-05 by davidhoness

jschrempp, 2016-03-14:
I used the evdev solution listed above with my SenseHat and it worked great. I never could get pygames to run headless.

themaire, 2016-04-03:
A big thanks for you tvoverbeek!!!

commit 5495cb31 (2015-10-05): thanks @tvoverbeek fixed #9