astro-pi/python-sense-hat #11: Incorporate joystick handling into the Sense HAT library

jrobinson-uk, 2015-09-28:
How easy would it be to add an evdev function to the existing sense-HAT library to abstract away all the keyboard handling code that users will have to include in their code if they want to get joystick events.

waveform80, 2015-09-28:
I've got a class that does this (part of stuff I was playing around with in #10). I could bung it all in another repo if anyone wants to take a look?

waveform80, 2015-09-28:
Oh, and I should add it doesn't depend on the python evdev library. I would've liked to have used that, but while it's available from PyPI, it's not packaged on Raspbian (I checked Jessie too) so I figured it was just easier to open the device directly and parse stuff from it (it's a trivial C struct after all).

XECDesign, 2015-09-28:
Yeah, avoiding the python evdev library would certainly be appreciated.

jrobinson-uk, 2015-09-28:
As long as it's simple and abstracts away some of details we'll be happy. @waveform80 pull request?

waveform80, 2015-09-28:
Sure, let me bung my working repo up on github and then see if I can merge some select bits with a clone of python-sense-hat

waveform80, 2015-09-28:
Okay, I've bunged it all in https://github.com/waveform80/pisense at the mo. Very rough - no README, no setup - this is all just me playing around with interfaces to the Sense HAT. The joystick stuff is in the SenseStick class in pisense/stick.py and can be used like this:

```python
>>> from pisense import SenseStick
>>> stick = SenseStick()
>>> stick.wait(timeout=10) # block (with timeout) until an event is available
True
>>> stick.read() # blocking read of one event
InputEvent(timestamp=1443450437.330044, key=106, state=1)
>>> stick.KEY_RIGHT
106
>>> stick.STATE_PRESS
1
```

It can also be used as an infinite iterator to read events:

```python
>>> d = {
...             stick.KEY_UP: 'up',
...             stick.KEY_DOWN: 'down',
...             stick.KEY_LEFT: 'left',
...             stick.KEY_RIGHT: 'right',
...             stick.KEY_ENTER: 'push',
...             }
>>> for event in stick:
...     if event.state == stick.STATE_PRESS:
...        print(d[event.key])
...
up
up
down
down
left
right
left
right
push
```

And as a context manager to handle implicitly closing the underlying file handle:

```python
>>> stick.close()
>>> with SenseStick() as stick:
...     stick.wait(10)
...
>>> # stick is now implicitly closed
```

I'll see what I can do about getting this incorporated into a clone of python-sense-hat in a mo. Would people prefer this in the SenseHAT class itself or left as a separate class?

N.B. The reason I've left it as a separate class in my experiment is that I think conflating all the functionality into a single class was a mistake: if you want to use the class as an iterator what is it providing? Readings from the joystick? Or one of the myriad sensors? Or all of them? Similar issues with context managers and so on.

waveform80, 2015-09-28:
Okay, there's the PR. NOTE: I haven't tested this at the moment (waiting for numpy to build in a venv on my A+!) but have a play with it and hopefully it'll work!

waveform80, 2015-09-28:
I've had a chance to test it now and it seems to work happily. Do let me know if the current structure's acceptable. If so, I'll get on with adding some proper documentation to it.

davidhoness, 2015-10-01:
I've merged the pull request from @waveform80

closed 2015-10-01 by davidhoness

ferrisbeuller, 2016-12-22:
Waveform80--thanks for this post!  i am new to Raspberry and this got me a lot more acquainted with it and Python.  I am still having issues with your class--or perhaps my undestanding everything still.

I cannot get the read() function to do anything.  it is like it is hanging on the device waiting.  Is there something i need to do to init the sense hat joystick?

have you done anymore work with this?

I simply cannot get the joystick to send me any events!

also, the built-in one does nothing for me either.  

Any help anyone?

waveform80, 2017-01-13:
@ferrisbeuller could you open a separate issue and include the code you're running and what you expect it to do? Sorry, I forgot about your comment and tend to skip looking at closed tickets!