astro-pi/python-sense-hat #13: Can't install pillow 3.2 to use Sense HAT

exidyboy, 2015-10-11:
Hi folks,

I am trying to set up 2 Sense HAT's and two new out of the box Pi 2 for a demo in 8 hours and am having massive issues with the pillow 3.2 install.

Hundreds of errors flowing off the screen. Have tried everything that I could Google up on different ways to install pillow but to no avail.

The last of the errors is (hand typed)

Command /usr/bin/python3.2 -c "import setuptools;_file"'/home/pibuild/pillow/setup.py';exec   blah blah blah with error code 1 in /home/pi/pillow

it says the error log is in /root/.pip/pip.log but I can't see that directory in root nor open it with nano

bennuttall, 2015-10-11:
Thanks for letting us know. I've seen someone else with the same error. Are you using the new Raspbian Jessie image? It should already be installed. Have you tried using it from Python?

@XECDesign doesn't the Jessie `sense-hat` package depend on `python3-pillow`? I thought the extra manual pillow install was only necessary in Wheezy? In which case we can remove it from the install instructions for Jessie users.

exidyboy, 2015-10-11:
I am using whatever noobs version you get when you buy a brand new Pi 2.
The Pi's just came off back order and four were couriered to me arriving the day before my flight overseas to give the talk.  I will check now - I have a couple of hours before I start setting up the room :-)

<--snip-->
pi@canterbury ~ $ cat /etc/*release
PRETTY_NAME="Raspbian GNU/Linux 7 (wheezy)"
NAME="Raspbian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=raspbian
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@canterbury ~ $

<--snip-->
So I am still running Wheezy. Can you confirm that Jessie would work out of the box. I might just have time to grab the image if I sit in the hotel foyer long enough.

The situation I believe at the moment is if you order a new Pi 2 with noobs and a Sense HAT and take them out of the box and do nothing else but:

sudo apt-get update
sudo apt-get upgrade (this step is not mentioned at https://www.raspberrypi.org/documentation/hardware/sense-hat/)
sudo apt-get install sense-hat
sudo pip-3.2 install pillow

You will be in a world of pain :-(

bennuttall, 2015-10-11:
Ah, so the Jessie image only just came out so the SD cards sold will have Wheezy on them for the next few weeks until they're updated.

The Jessie image you can grab from [raspberrypi.org](https://www.raspberrypi.org/downloads/raspbian/) has Sense HAT software pre-installed.

exidyboy, 2015-10-11:
Thanks Ben,
I have 45 minutes to go on the Jessie download.
Is the best resource for some visually compelling demo scripts:
https://www.raspberrypi.org/learning/getting-started-with-the-sense-hat/worksheet/

bennuttall, 2015-10-11:
Try these:
- https://github.com/RPi-Distro/python-sense-hat/tree/master/examples
- https://github.com/bennuttall/sense-hat-examples

exidyboy, 2015-10-13:
Hi Ben,

I've downloaded the current Jessie .zip from https://www.raspberrypi.org/downloads/raspbian/ several times but the unzipped .img has failed checksum verification each time. Thankyou for your timely assistance and I look forward to getting the Sense HAT's running when I get home.

Cheers,
Mike

bennuttall, 2015-10-13:
The checksum on the website is of the zip - not the extracted img. Have you tried writing it to a card?

exidyboy, 2015-10-13:
Hi Ben,
Just wrote a card and have it working. I'm going to image another card and lug everything down to the hotel lobby to connect to wired Ethernet and download the sample programs you linked to above. Unfortunately the fact that the Epiphany browser and Wi-Pi do not play well with public Wi-Fi captive portal platforms has been frustrating many of us over the last few days.

Thanks again,
Mike

closed 2015-10-13 by exidyboy

DaddyTheRunner, 2015-10-15:
So is there a solution for Wheezy?  I can't get the following command to work under Wheezy:

`sudo pip-3.2 install pillow`

I get the following error message:

`ValueError: --enable-jpeg requested but jpeg not found, aborting.

Any tips other than upgrading to Jessie?

PiInTheSky, 2015-11-13:
Do this:

sudo apt-get install libjpeg-dev

then re-install pillow.

cliffordkasper, 2015-12-25:
The sudo apt-get install libjpeg-dev worked for me, thanks!