astro-pi/python-sense-hat #21: Documentation on dependencies

5chdn, 2015-11-28:
Hi, where can I find a documentation on dependencies?

I'm tryint to install sense-hat on archlinux and i struggle to figure out what's missing.

What is `import RTIMU  # custom version` for example?

XECDesign, 2015-11-29:
The RTIMULib changes were upstreamed, so it's not really a custom version. However, RTIMULib has been abandoned, so I am keeping this as 'the' version used https://github.com/RPi-Distro/RTIMULib. It may be modified in the future, but for now should be identical to the regular RTIMULib. Other dependencies that I know of are pillow and numpy.

5chdn, 2015-11-29:
Okay, thanks. I was able to get this running on [ArchlinuxARM](http://archlinuxarm.org/platforms/armv7/broadcom/raspberry-pi-2) and submitted packages to the [Arch User Repository](https://wiki.archlinux.org/index.php/Arch_User_Repository) for `armv7l`:

- https://aur.archlinux.org/packages/python-sense-hat/
- https://aur.archlinux.org/packages/python-rtimulib/

For the documentation you could add something like that:

```
pacaur -Syu
pacaur -S python-sense-hat
```

And maybe list the dependencies somewhere if anybody likes to build it from source? :-)

5chdn, 2015-11-30:
what ever. keep it simple and nobody will figure how to use.

closed 2015-11-30 by 5chdn