astro-pi/python-sense-hat #24: Python3 gives inconsistent results for raw accelerometer readings

topshed, 2015-12-11:
Reading from the accelerometer in Python3 gives inconsistent results when using get_accelerometer_raw().values() 

If you  repeatedly run this simple code:

from sense_hat import SenseHat
sh = SenseHat()
print(sh.get_accelerometer_raw().values()

and leave the PI/SenseHAt in the same position, the output will be significantly different each time. If you try to round() the x,y,z values to determine the Pi's orientation, it will give different results (sometimes values will round to 0, sometimes to 1).

Running the same code with Python2.7 always gives consistent values:

[image]

I've tried this on two different Pis with 2 different SenseHATs. Latest Jessie build with latest sense-hat libraries.

lpashwin, 2015-12-11:
Well this is because, when you use *get_accelerometer_raw().values()* you are getting the key values of the dictionary *get_accelerometer_raw()*. This need not be ordered in Python3. I think they used to be ordered in python2.x, though I have a faint recollection of running into trouble with dictionary value ordering in python2.x. 

This is how I would make sure that the order is maintained 
print([get_accelerometer_raw()[key] for key in ['x','y','z']])

assuming the key values pf accelerometer are x,y,z, you can check it by running get_accelerometer_raw().keys() once. *I am sorry I am away from my Pi to check it*

And  as always there might be better ways to do it!

topshed, 2015-12-12:
Yes, that works correctly, thanks. I didn't read the docs carefully enough -  we were following the "detecting movement" example described here https://www.raspberrypi.org/learning/getting-started-with-the-sense-hat/worksheet/
which says to use Python3 but will only work for Python2.

closed 2016-06-06 by bennuttall