astro-pi/python-sense-hat #25: Initializing environmental sensors failed

5chdn, 2015-12-21:
Initializing environmental sensors failed. 

```
0 root@rpi /home/user/src $ whoami
root
0 root@rpi /home/user/src $ python --version
Python 3.5.1
0 root@rpi /home/user/src $ python ./temp.py 
Traceback (most recent call last):
  File "./temp.py", line 165, in <module>
    print(s.get_temperature())
  File "/usr/lib/python3.5/site-packages/sense_hat/sense_hat.py", line 591, in get_temperature
    return self.get_temperature_from_humidity()
  File "/usr/lib/python3.5/site-packages/sense_hat/sense_hat.py", line 567, in get_temperature_from_humidity
    self._init_humidity()  # Ensure humidity sensor is initialised
  File "/usr/lib/python3.5/site-packages/sense_hat/sense_hat.py", line 534, in _init_humidity
    raise OSError('Humidity Init Failed, please run as root / use sudo')
OSError: Humidity Init Failed, please run as root / use sudo
1 root@rpi /home/user/src $ uname -a
Linux rpi 4.1.13-1-ARCH #1 SMP PREEMPT Fri Nov 13 20:11:44 MST 2015 armv7l GNU/Linux
```

I am root already. All of the environmental API calls fail with the same error. Both pressure and humidity sensors are affected.

The LED matrix is working fine. Running ArchLinuxARM.

XECDesign, 2015-12-22:
Is i2c enabled (including the i2c-dev module). Do the /dev nodes show up? What does `i2cdetect -y 1` say?

5chdn, 2015-12-22:
Ok that was missing. Installing the module solved this issue. Thanks.
For future reference: https://wiki.archlinux.org/index.php/Raspberry_Pi#I2C

closed 2015-12-22 by 5chdn

jwhendy, 2017-09-24:
Leaving breadcrumbs. In this case the module wasn't loaded. For #79 , which gives a very similar error, it was the permissions of `/dev/i2c-1` causing the issue.