astro-pi/python-sense-hat #26: OSError: Cannot detect RPi-Sense FB device on RASPBIAN JESSIE

amnon409, 2015-12-22:
Hello, 

I having the following error while using the last OSError: Cannot detect RPi-Sense FB device
RASPBIAN JESSIE  image release 2015-11-21
I tried to enabled device tree in raspi-config which did not help 
When boot all led on the Sense-hat are on 

Thanks 

Amnon

XECDesign, 2015-12-22:
Is this with a clean image or have changes been made? Device tree would have already been enabled and everything should work without having to make any additional changes. The expected behaviour is that when power is applied, all LEDs light up and are turned off once the relevant kernel module loads.

amnon409, 2015-12-22:
Yes I used a clean image ( last one ) no change made since per my
understanding all should be already set and work.

When I got the Sense_hat ( 2-3 month ago) I used a different image which
was available at that time not sure which version it was but I had to
install the Sense_Hat  on it
The light went off once you boot. and it was working.

Any help will be appropriate

XECDesign, 2015-12-22:
Could you please run `sudo apt-get install device-tree-compiler` then provide the output of `dmesg` and `dtc -I fs /proc/device-tree`?

amnon409, 2015-12-22:
I re-image again the card.
I believe the problem I had before is that I did not expend the disk to the
max size
It is working now :-)   ,

Thank  you !!!!

XECDesign, 2015-12-22:
No worries, great to hear it's working fine now.

closed 2015-12-22 by XECDesign