astro-pi/python-sense-hat #28: OSError: Cannot detect RPi-Sense FB device

El-Don-Quijote, 2015-12-31:
hello there,
using my new sensehat on my raspberry pi b+ with a newly installed raspbian via NOOBS i get the following error message

OSError: Cannot detect RPi-Sense FB device

altough all LEDs are on. Similiar to following issue (https://github.com/RPi-Distro/python-sense-hat/issues/8), i also enabled the device tree, however, nothing changed.

i would be very glad for any kind of help!
and a happy new year! :)

XECDesign, 2016-01-11:
Hello, could you please provide the output of the following:
```
sudo apt-get install device-tree-compiler
uname -a
vcgencmd version
sudo dtc -I fs /proc/device-tree
```

ptrelford, 2016-01-13:
I've been experiencing the same error when trying to run the python samples, i.e. "Cannot detect RPi-Sense FB device"

I've tried plugging the Sense Hat into 2 different Raspberry Pi 2 devices, and I've tried both Raspbian Wheezy and a fresh install of Jessie, but still keep coming up against the same issue.

For Wheezy I followed all the additional steps I've managed to find scattered around the internet:
 - [Can't install pillow 3.2 to use Sense HAT](https://github.com/RPi-Distro/python-sense-hat/issues/13) "jpeg not" found issue
   - sudo apt-get install libjpeg-dev (see [Installation step missing for Wheezy](https://github.com/RPi-Distro/python-sense-hat/issues/17))
 - enabling the dev-tree
   - sudo raspi-config and turning it on in Advanced options
 - enabling i2c
   - sudo raspi-config and turning it on in Advanced options

Please help :)

XECDesign, 2016-01-13:
Hello, could you please provide the output of the following:
```
sudo apt-get install device-tree-compiler
uname -a
vcgencmd version
sudo dtc -I fs /proc/device-tree
dmesg
ls /devfb*
```
Please use a clean jessie with no modifications, wheezy is not supported.

eichin, 2016-01-14:
I don't know if this is quite relevant, but through some discussion on reddit, I had a SenseHat on a Pi Zero running Jessie-Lite, and using `raspi-config` to enable i2c got me the sensors but no LED activity; installing and running `rpi-update` (and rebooting) did, however, enable the LEDs (based on `vcgencmd version`, going from Nov 18 2015 15:43:45/0c9af7cde38e223c95c3404f468db8fc5004495c where they didn't work, to Jan 13 2016 20:33:52/17c0f609ee91eb69090c22a3e0e8dc907b696096 where they did.)

recantha, 2016-01-15:
Try adding dtoverlay=rpi-sense to the bottom of /boot/config.txt
That worked for me.

AlainGeenrits, 2016-01-22:
I am using the sense hat on Pi2 and tried everything. I had device - tree enabled. This last comment by recantha solved it for me ! A million thx ! Why is this not in the doc ??

XECDesign, 2016-01-22:
Because this shouldn't be a required step. I keep asking the same questions to try to figure out why this is happening to some people, but am getting ignored.

AlainGeenrits, 2016-01-22:
I am sorry to hear. How can I help ? Do you need output from my system to help you ?

XECDesign, 2016-01-23:
It would be very helpful if you could remove that line from config.txt, install device-tree-compiler, reboot and then provide the output of the following commands:
```
uname -a
vcgencmd version
sudo dtc -I fs /proc/device-tree
dmesg
ls /dev/fb*
```

AlainGeenrits, 2016-01-26:
Here you go ! Hope it helps !

uame -a : Linux raspberrypi 4.1.13-v7+ #826 SMP PREEMPT Fri Nov 13 20:19:03
GMT 2015 armv7l GNU/Linux

vcgencmd version
Nov 18 2015 15:43:45
Copyright (c) 2012 Broadcom
version 0c9af7cde38e223c95c3404f468db8fc5004495c (clean) (release)

ls /dev/fb*
/dev/fb0

device-tree and dmesg output attached as text files.

See you on the bitstream, Alain

-- 
See you on the bitstream, Alain

Alain Geenrits - alain@geenrits.net
twitter: alaingeenrits
blog: http://bitstream.geenrits.net

/dts-v1/;

/ {
	model = "Raspberry Pi 2 Model B Rev 1.1";
	compatible = "brcm,bcm2709";
	memreserve = <0x3b000000 0x4000000>;
	#address-cells = <0x1>;
	#size-cells = <0x1>;
	interrupt-parent = <0x1>;

	axi {

		dma {
			broadcom,channels = <0x7f35>;
		};

		usb {

			hub {

				ethernet {
					mac-address = [b8 27 eb b0 f2 bf];
				};
			};
		};

		sdhci {
			clock-frequency = <0xee6b280>;
		};

		uart0 {
			clock-frequency = <0x2dc6c0>;
		};

		vc_mem {
			reg = <0x3dc00000 0x3f000000 0xc0000000>;
		};
	};

	soc {
		compatible = "simple-bus";
		ranges = <0x7e000000 0x3f000000 0x1000000>;
		#address-cells = <0x1>;
		phandle = <0x22>;
		#size-cells = <0x1>;
		linux,phandle = <0x22>;
[... 898 lines cut ...]
[    6.256759] spi spi0.0: setting up native-CS0 as GPIO 8
[    6.257754] spi spi0.1: setting up native-CS1 as GPIO 7
[    6.259822] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    6.294771] random: nonblocking pool is initialized
[    6.756775] usbcore: registered new interface driver rtl8192cu
[    7.163145] systemd-journald[109]: Received request to flush runtime journal from PID 1
[    8.157982] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    8.782463] cfg80211: Calling CRDA to update world regulatory domain
[    8.890484] cfg80211: World regulatory domain updated:
[    8.890525] cfg80211:  DFS Master region: unset
[    8.890540] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[    8.890559] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    8.890576] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[    8.890592] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[    8.890615] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[    8.890643] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[    8.890660] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[    8.890676] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[    8.890692] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[    9.066123] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:135164k SSFS
[   10.334405] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   12.607330] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   12.607923] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   13.284670] bcm2708_fb soc:fb: Unknown ioctl 0x40187a22
[   22.553831] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)

XECDesign, 2016-01-26:
Thanks, that's very helpful. It looks like the auto-detection doesn't work.
Could you add `dtdebug=1` to `/boot/config.txt`, reboot and then provide the output of `sudo vcdbg log msg 2>&1`

AlainGeenrits, 2016-01-27:
Here you go !

-- 
See you on the bitstream, Alain

Alain Geenrits - alain@geenrits.net
twitter: alaingeenrits
blog: http://bitstream.geenrits.net

000763.271: HDMI:EDID version 1.3, 1 extensions, screen size 48x27 cm
000763.336: HDMI:EDID features - videodef 0x80 !standby !suspend active off; colour encoding:RGB444|YCbCr422; sRGB is not default colourspace; preferred format is native; does not support GTF
000763.396: HDMI:EDID found preferred CEA detail timing format: 1920x1080p @ 60 Hz (16)
000763.436: HDMI:EDID found DMT format: code 4, 640x480p @ 60 Hz in established timing I/II
000763.475: HDMI:EDID found DMT format: code 5, 640x480p @ 72 Hz in established timing I/II
000763.512: HDMI:EDID found DMT format: code 6, 640x480p @ 75 Hz in established timing I/II
000763.551: HDMI:EDID found DMT format: code 9, 800x600p @ 60 Hz in established timing I/II
000763.588: HDMI:EDID found DMT format: code 11, 800x600p @ 75 Hz in established timing I/II
000763.628: HDMI:EDID found DMT format: code 16, 1024x768p @ 60 Hz in established timing I/II
000763.667: HDMI:EDID found DMT format: code 18, 1024x768p @ 75 Hz in established timing I/II
000763.707: HDMI:EDID found DMT format: code 36, 1280x1024p @ 75 Hz in established timing I/II
000763.795: HDMI:EDID standard timings block x 8: 0xD1C0 9500 950F B300 8180 81C0 0101 0101 
000763.859: HDMI:EDID found DMT format: code 82, 1920x1080p @ 60 Hz (16:9) in standard timing 0
000763.912: HDMI:EDID found DMT format: code 47, 1440x900p @ 60 Hz (16:10) in standard timing 1
000763.962: HDMI:EDID found DMT format: code 48, 1440x900p @ 75 Hz (16:10) in standard timing 2
000764.015: HDMI:EDID found DMT format: code 58, 1680x1050p @ 60 Hz (16:10) in standard timing 3
000764.065: HDMI:EDID found DMT format: code 35, 1280x1024p @ 60 Hz (5:4) in standard timing 4
000764.117: HDMI:EDID found DMT format: code 85, 1280x720p @ 60 Hz (16:9) in standard timing 5
000777.127: HDMI:EDID parsing v3 CEA extension 0
000777.164: HDMI:EDID monitor support - underscan IT formats:yes, basic audio:yes, yuv444:yes, yuv422:yes, #native DTD:1
000777.206: HDMI:EDID found CEA detail timing format: 1920x1080p @ 60 Hz (16)
000777.245: HDMI:EDID found CEA detail timing format: 720x480p @ 60 Hz (2)
000777.288: HDMI:EDID found CEA detail timing format: 1280x720p @ 60 Hz (4)
000777.332: HDMI:EDID found CEA detail timing format: 720x576p @ 50 Hz (17)
000777.366: HDMI:EDID found CEA format: code 1, 640x480p @ 60Hz 
000777.399: HDMI:EDID found CEA format: code 2, 720x480p @ 60Hz 
000777.432: HDMI:EDID found CEA format: code 3, 720x480p @ 60Hz 
000777.465: HDMI:EDID found CEA format: code 5, 1920x1080i @ 60Hz 
000777.498: HDMI:EDID found CEA format: code 6, 1440x480i @ 60Hz 
000777.531: HDMI:EDID found CEA format: code 7, 1440x480i @ 60Hz 
000777.562: HDMI:EDID found CEA format: code 16, 1920x1080p @ 60Hz 
000777.596: HDMI:EDID found CEA format: code 17, 720x576p @ 50Hz 
000777.627: HDMI:EDID found CEA format: code 18, 720x576p @ 50Hz 
000777.660: HDMI:EDID found CEA format: code 19, 1280x720p @ 50Hz 
000777.693: HDMI:EDID found CEA format: code 20, 1920x1080i @ 50Hz 
000777.725: HDMI:EDID found CEA format: code 21, 1440x576i @ 50Hz 
000777.759: HDMI:EDID found CEA format: code 22, 1440x576i @ 50Hz 
000777.793: HDMI:EDID found CEA format: code 31, 1920x1080p @ 50Hz 
000777.826: HDMI:EDID found CEA format: code 4, 1280x720p @ 60Hz 
000777.882: HDMI:EDID found audio format 2 channels PCM, sample rate: 32|44|48|96 kHz, sample size: 16|20|24 bits
000777.901: HDMI:EDID found HDMI VSDB length 5
000777.925: HDMI:EDID HDMI VSDB has physical address 1.0.0.0
000777.940: HDMI:EDID HDMI VSDB has no extension fields
000777.982: HDMI:EDID filtering formats with pixel clock > 162 MHz or h. blanking > 1023
000778.167: HDMI:EDID preferred mode remained as CEA (16) 1920x1080p @ 60 Hz with pixel clock 148 MHz
000782.322: HDMI:Setting property pixel encoding to Default
000782.342: HDMI:Setting property pixel clock type to PAL
000782.363: HDMI:Setting property content type flag to No data
000782.381: HDMI:Setting property fuzzy format match to enabled
000797.587: gpioman: gpioman_get_pin_num: pin LEDS_RUNNING not defined
000797.613: gpioman: gpioman_get_pin_num: pin LEDS_NAND_ACTIVITY not defined
000797.636: gpioman: gpioman_get_pin_num: pin LEDS_USB_ACTIVITY not defined
000797.658: gpioman: gpioman_get_pin_num: pin LEDS_FATAL_ERROR not defined
000797.678: gpioman: gpioman_get_pin_num: pin LEDS_APP_OK not defined
000797.699: gpioman: gpioman_get_pin_num: pin LEDS_APP_FAILED not defined
000797.721: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_AUTH not defined
000797.742: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_UNAUTH not defined
000797.761: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_ON not defined
000797.782: gpioman: gpioman_get_pin_num: pin LEDS_DVI_ON not defined
000797.802: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_HPD_UP not defined
000797.825: gpioman: gpioman_get_pin_num: pin LEDS_REMOTE_CONTROL not defined
[... 86 lines cut ...]
001464.749: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
001470.552: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
001476.378: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
001482.288: dtdebug:   override arm_freq: cell target clock-frequency @ offset 0 (size 4)
001488.262: dtparam: cache_line_size=64
001489.781: dtdebug: Found override cache_line_size
[... cut ...]
003307.424: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_UNAUTH not defined
003307.447: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_ON not defined
003307.469: gpioman: gpioman_get_pin_num: pin LEDS_DVI_ON not defined
003307.492: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_HPD_UP not defined
003307.513: gpioman: gpioman_get_pin_num: pin LEDS_REMOTE_CONTROL not defined
003307.535: gpioman: gpioman_get_pin_num: pin LEDS_ARM_CONTROLLED not defined
003307.927: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010

XECDesign, 2016-01-27:
Thanks again. Looks like either a firmware or hardware fault. Tried my sense hat and it works as expected. I've seen someone report that they've had a few not working in a classroom, which seems to indicate it's a hardware issue, specifically with the eeprom.

AlainGeenrits, 2016-01-27:
You are welcome ! I will try another one to check. This is an RPi2.

XECDesign, 2016-01-27:
To clarify, this would be an issue with the HAT, not the pi.

I will add instructions on how to program the EEPROM to fix this later.

XECDesign, 2016-02-02:
One final bit of information is needed to confirm whether the EEPROM is the problem.
Could someone please read the eeprom using [these instructions here](https://github.com/raspberrypi/documentation/blob/master/hardware/sense-hat/README.md#eeprom-data) and then send me sense_read.eep?
Either a link to where it's uploaded or an email attachment would be great (serge at the obvious address dot org).

AlainGeenrits, 2016-02-05:
Here is the file I obtained attached. Hope it helps !

XECDesign, 2016-02-05:
I've got the file and can confirm that the EEPROM data is incomplete. The issue can be resolved using flashing the eeprom data yourself, but I recommend saving the old data first. The device UUID from the old data may help the factory track down the issue.

Thanks again for the help.

denizozger, 2016-02-10:
Hi @XECDesign, unfortunately I have the same issue. It's a Pi Zero, when I turn it on all LEDs are momentarily lights up, but I can't run any of the example SenseHAT code.

```
pi@raspberrypi:~ $ python /usr/src/sense-hat/examples/python-sense-hat/text_scroll.py
Traceback (most recent call last):
  File "/usr/src/sense-hat/examples/python-sense-hat/text_scroll.py", line 4, in <module>
    sense = SenseHat()
  File "/usr/lib/python2.7/dist-packages/sense_hat/sense_hat.py", line 36, in __init__
    raise OSError('Cannot detect %s device' % self.SENSE_HAT_FB_NAME)
OSError: Cannot detect RPi-Sense FB device
```

I have

- Enabled i2c
- Enabled dev-tree
- Tried to add ```dtdebug=1``` to ```/boot/config.txt``` but no luck

Output of the following commands are below:
```
uname -a
vcgencmd version
sudo dtc -I fs /proc/device-tree
dmesg
ls /dev/fb*
```

> pi@raspberrypi:~ $ uname -a
> Linux raspberrypi 4.1.17+ #838 Tue Feb 9 12:57:10 GMT 2016 armv6l GNU/Linux
> pi@raspberrypi:~ $ vcgencmd version
> Feb  9 2016 19:35:19
> Copyright (c) 2012 Broadcom
> version dc2c1aba32c7469e82a87be0d5d921aa9233e60d (clean) (release)
> pi@raspberrypi:~ $ sudo dtc -I fs /proc/device-tree
> Warning (avoid_default_addr_size): Relying on default #address-cells value for /axi/vc_mem
> Warning (avoid_default_addr_size): Relying on default #size-cells value for /axi/vc_mem
> /dts-v1/;
> 
> / {
> 	model = "Raspberry Pi Zero Rev 1.2";
> 	compatible = "brcm,bcm2708";
> 	memreserve = <0x1c000000 0x4000000>;
> 	#address-cells = <0x1>;
> 	#size-cells = <0x1>;
> 	interrupt-parent = <0x1>;
> 
> 	axi {
> 
> 		dma {
> 			broadcom,channels = <0x7f35>;
> 		};
> 
> 		usb {
> 
> 			hub {
> 
> 				ethernet {
> 					mac-address = [b8 27 eb 91 64 27];
> 				};
> 			};
> 		};
> 
> 		vc_mem {
> 			reg = <0x1ec00000 0x20000000 0x40000000>;
> 		};
> 	};
> 
> 	soc {
> 		compatible = "simple-bus";
> 		ranges = <0x7e000000 0x20000000 0x1000000>;
[... 811 lines cut ...]
> [   12.853563] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [   12.853576] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [   12.853603] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
> [   12.853616] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
> [   12.853629] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
> [   12.853640] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
> [   12.853651] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
> [   12.853661] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
> [   13.863789] brcmfmac: brcmf_add_if: ERROR: netdev:wlan0 already exists
> [   13.863818] brcmfmac: brcmf_add_if: ignore IF event
> [   14.979484] cfg80211: Calling CRDA for country: GB
> [   15.067511] cfg80211: Regulatory domain changed to country: GB
> [   15.067560] cfg80211:  DFS Master region: ETSI
> [   15.067570] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
> [   15.067585] cfg80211:   (2402000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
> [   15.067599] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
> [   15.067611] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
> [   15.067623] cfg80211:   (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
> [   15.067648] cfg80211:   (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
> [   18.165362] Adding 102396k swap on /var/swap.  Priority:-1 extents:2 across:839676k SSFS
> [   34.701889] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
> pi@raspberrypi:~ $ ls /dev/fb*
> /dev/fb0

Thanks!

XECDesign, 2016-02-10:
@denizozger , I would recommend recantha's solution (adding dtoverlay=rpi-sense to the bottom of /boot/config.txt). If that fixes it, you can consider flashing the eeprom yourself to make sure it just work without needing to adjust config.txt in the future.

denizozger, 2016-02-10:
I added ```dtoverlay=rpi-sense``` to the bottom of ```/boot/config.txt```, and rebooted, still getting the same error.

Any ideas? Is there are way to double check if ```dtoverlay``` is honoured?

XECDesign, 2016-02-10:
You should be able to tell from the output of `sudo vcdbg log msg` and the dtc command from before.

denizozger, 2016-02-11:
The output of ```sudo vcdbg log msg``` and ```sudo dtc -I fs /proc/device-tree``` are below. There's nothing about ```dtoverlay ```, although I can confirm the last line of ```/boot/config.txt``` is ```dtoverlay=rpi-sense```.

> pi@raspberrypi:~ $ sudo vcdbg log msg
> 000944.496: clock: clock_set_clock:  dst: CLK_H264, source: XOSC, div: 0.000000
> 000944.532: clock: clock_set_clock:  dst: CLK_ISP, source: XOSC, div: 0.000000
> 000944.570: clock: clock_set_clock:  dst: CLK_H264, source: XOSC, div: 0.000000
> 000944.606: clock: clock_set_clock:  dst: CLK_ISP, source: XOSC, div: 0.000000
> 000944.646: clock: clock_set_clock:  dst: CLK_VEC, source: XOSC, div: 1.000000
> 000944.679: clock: clock_set_clock:  dst: CLK_VEC, source: XOSC, div: 0.000000
> 000944.972: HDMI:EDID error reading EDID block 0 attempt 0
> 000946.236: HDMI:EDID error reading EDID block 0 attempt 1
> 000947.494: HDMI:EDID error reading EDID block 0 attempt 2
> 000948.752: HDMI:EDID error reading EDID block 0 attempt 3
> 000950.011: HDMI:EDID error reading EDID block 0 attempt 4
> 000951.269: HDMI:EDID error reading EDID block 0 attempt 5
> 000952.526: HDMI:EDID error reading EDID block 0 attempt 6
> 000953.785: HDMI:EDID error reading EDID block 0 attempt 7
> 000955.043: HDMI:EDID error reading EDID block 0 attempt 8
> 000956.302: HDMI:EDID error reading EDID block 0 attempt 9
> 000957.323: HDMI:EDID giving up on reading EDID block 0
> 000957.382: clock: clock_set_clock:  dst: CLK_VEC, source: XOSC, div: 1.000000
> 000957.415: clock: clock_set_clock:  dst: CLK_VEC, source: XOSC, div: 0.000000
> 000959.985: HDMI:Setting property pixel encoding to Default
> 000960.005: HDMI:Setting property pixel clock type to PAL
> 000960.023: HDMI:Setting property content type flag to No data
> 000960.043: HDMI:Setting property fuzzy format match to enabled
> 000975.432: gpioman: gpioman_get_pin_num: pin LEDS_RUNNING not defined
> 000975.456: gpioman: gpioman_get_pin_num: pin LEDS_NAND_ACTIVITY not defined
> 000975.477: gpioman: gpioman_get_pin_num: pin LEDS_USB_ACTIVITY not defined
> 000975.497: gpioman: gpioman_get_pin_num: pin LEDS_FATAL_ERROR not defined
> 000975.519: gpioman: gpioman_get_pin_num: pin LEDS_APP_OK not defined
> 000975.541: gpioman: gpioman_get_pin_num: pin LEDS_APP_FAILED not defined
> 000975.562: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_AUTH not defined
> 000975.584: gpioman: gpioman_get_pin_num: pin LEDS_HDCP_UNAUTH not defined
> 000975.606: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_ON not defined
> 000975.628: gpioman: gpioman_get_pin_num: pin LEDS_DVI_ON not defined
> 000975.650: gpioman: gpioman_get_pin_num: pin LEDS_HDMI_HPD_UP not defined
> 000975.670: gpioman: gpioman_get_pin_num: pin LEDS_REMOTE_CONTROL not defined
> 000975.691: gpioman: gpioman_get_pin_num: pin LEDS_ARM_CONTROLLED not defined
> 000975.797: clock: Set PLLB_VCO to 640000000
> 000975.910: clock: clock_set_clock:  dst: CLK_ARM, source: PLLB_ARM, div: 1.000000
> 000975.945: clock: clock_set_clock:  dst: CLK_ARM, source: XOSC, div: 0.000000
> 000975.982: clock: clock_set_clock:  dst: CLK_ARM, source: XOSC, div: 1.000000
> 000976.014: clock: clock_set_clock:  dst: CLK_ARM, source: XOSC, div: 0.000000
> 000976.039: clock: Set PLLB_VCO to 640000000
> 000976.149: clock: clock_set_clock:  dst: CLK_ARM, source: PLLB_ARM, div: 1.000000
> 000976.177: clock: Set PLLB_VCO to 1200000000
> 000976.249: clock: clock_set_clock:  dst: CLK_ARM, source: PLLB_ARM, div: 1.000000
> 000993.396: clock: clock_set_clock:  dst: CLK_ARM, source: XOSC, div: 0.000000
> 000998.023: hdmi: HDMI:>>>>>>>>>>>>>Rx sensed, reading EDID<<<<<<<<<<<<<
> 000998.103: clock: clock_set_clock:  dst: CLK_H264, source: XOSC, div: 0.000000
> 000998.142: clock: clock_set_clock:  dst: CLK_ISP, source: XOSC, div: 0.000000
> 000998.178: clock: clock_set_clock:  dst: CLK_H264, source: XOSC, div: 0.000000
> 000998.211: clock: clock_set_clock:  dst: CLK_ISP, source: XOSC, div: 0.000000
> 000998.254: clock: clock_set_clock:  dst: CLK_VEC, source: XOSC, div: 1.000000
> 000998.287: clock: clock_set_clock:  dst: CLK_VEC, source: XOSC, div: 0.000000
> 000998.597: hdmi: HDMI:EDID error reading EDID block 0 attempt 0
> 000999.866: hdmi: HDMI:EDID error reading EDID block 0 attempt 1
> 001001.130: hdmi: HDMI:EDID error reading EDID block 0 attempt 2
> 001002.394: hdmi: HDMI:EDID error reading EDID block 0 attempt 3
> 001003.657: hdmi: HDMI:EDID error reading EDID block 0 attempt 4
> 001004.921: hdmi: HDMI:EDID error reading EDID block 0 attempt 5
> 001006.184: hdmi: HDMI:EDID error reading EDID block 0 attempt 6
> 001007.447: hdmi: HDMI:EDID error reading EDID block 0 attempt 7
> 001008.710: hdmi: HDMI:EDID error reading EDID block 0 attempt 8
> 001009.973: hdmi: HDMI:EDID error reading EDID block 0 attempt 9
> 001010.999: hdmi: HDMI:EDID giving up on reading EDID block 0
> 001011.029: hdmi: HDMI: No lookup table for resolution group 0
> 001011.046: hdmi: HDMI: hotplug attached with DVI support
> 001011.097: clock: clock_set_clock:  dst: CLK_VEC, source: XOSC, div: 1.000000
[... 806 lines cut ...]
> 		i2c0 = [00 00 00 15 73 74 61 74 75 73 00];
> 		i2c1 = [00 00 00 16 73 74 61 74 75 73 00];
> 		i2c_arm = [00 00 00 16 73 74 61 74 75 73 00];
> 		audio = [00 00 00 1a 73 74 61 74 75 73 00];
> 		uart0 = [00 00 00 12 73 74 61 74 75 73 00];
> 		i2c2_iknowwhatimdoing = [00 00 00 17 73 74 61 74 75 73 00];
> 		act_led_activelow = <0x18 0x6770696f 0x733a3800>;
> 		pwr_led_gpio = <0x19 0x6770696f 0x733a3400>;
[... cut ...]
> 		random = [00 00 00 1c 73 74 61 74 75 73 00];
> 		i2c2_baudrate = [00 00 00 17 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
> 		core_freq = [00 00 00 06 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
> 		i2c_arm_baudrate = [00 00 00 16 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
> 		pwr_led_activelow = <0x19 0x6770696f 0x733a3800>;
> 		act_led_gpio = <0x18 0x6770696f 0x733a3400>;
> 		cache_line_size = [00 00 00 11 63 61 63 68 65 2d 6c 69 6e 65 2d 73 69 7a 65 3a 30 00];
> 	};
> };

XECDesign, 2016-02-11:
Looks like the overlay is loaded (`001336.702: Loaded overlay 'rpi-sense'`) and the device tree data is good too.

What's the output of dmesg?

AlainGeenrits, 2016-02-11:
Silly question, but where would I find the correct image to flash ? Is it on github ?

XECDesign, 2016-02-11:
@AlainGeenrits, the data is created when do the first step in the 'writing' section [here](https://github.com/raspberrypi/documentation/blob/master/hardware/sense-hat/README.md#eeprom-data).

denizozger, 2016-02-11:
Output of ```dmesg```:

```
pi@raspberrypi:~ $ dmesg
[ 0.000000] Booting Linux on physical CPU 0x0
[ 0.000000] Initializing cgroup subsys cpuset
[ 0.000000] Initializing cgroup subsys cpu
[ 0.000000] Initializing cgroup subsys cpuacct
[ 0.000000] Linux version 4.1.17+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #838 Tue Feb 9 12:57:10 GMT 2016
[ 0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[ 0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[ 0.000000] Machine model: Raspberry Pi Zero Rev 1.2
[ 0.000000] cma: Reserved 8 MiB at 0x1b400000
[ 0.000000] Memory policy: Data cache writeback
[ 0.000000] On node 0 totalpages: 114688
[ 0.000000] free_area_init_node: node 0, pgdat c0833994, node_mem_map db010000
[ 0.000000] Normal zone: 1008 pages used for memmap
[ 0.000000] Normal zone: 0 pages reserved
[ 0.000000] Normal zone: 114688 pages, LIFO batch:31
[ 0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=132768
[ 0.000000] pcpu-alloc: [0] 0
[ 0.000000] Built 1 zonelists in Zone order, mobility grouping on. Total pages: 113680
[ 0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=592 bcm2708_fb.fbheight=448 bcm2708.boardrev=0x900092 bcm2708.serial=0xde916427 smsc95xx.macaddr=B8:27:EB:91:64:27 bcm2708_fb.fbswap=1 bcm2708.uart_clock=3000000 bcm2708.disk_led_gpio=47 sdhci-bcm2708.emmc_clock_freq=250000000 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000 dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[ 0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[ 0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[ 0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[ 0.000000] Memory: 436860K/458752K available (5771K kernel code, 488K rwdata, 1788K rodata, 348K init, 720K bss, 13700K reserved, 8192K cma-reserved)
[ 0.000000] Virtual kernel memory layout:
vector : 0xffff0000 - 0xffff1000 ( 4 kB)
fixmap : 0xffc00000 - 0xfff00000 (3072 kB)
vmalloc : 0xdc800000 - 0xff000000 ( 552 MB)
lowmem : 0xc0000000 - 0xdc000000 ( 448 MB)
modules : 0xbf000000 - 0xc0000000 ( 16 MB)
.text : 0xc0008000 - 0xc076a0cc (7561 kB)
.init : 0xc076b000 - 0xc07c2000 ( 348 kB)
.data : 0xc07c2000 - 0xc083c278 ( 489 kB)
.bss : 0xc083c278 - 0xc08f03e0 ( 721 kB)
[ 0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[ 0.000000] NR_IRQS:522
[ 0.000000] clocksource stc: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[ 0.000013] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[ 0.000053] Switching to timer-based delay loop, resolution 1000ns
[ 0.000334] Console: colour dummy device 80x30
[ 0.001268] console [tty1] enabled
[ 0.001324] Calibrating delay loop (skipped), value calculated using timer frequency.. 2.00 BogoMIPS (lpj=10000)
[ 0.001403] pid_max: default: 32768 minimum: 301
[ 0.001767] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[ 0.001833] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[ 0.002882] Initializing cgroup subsys blkio
[ 0.002969] Initializing cgroup subsys memory
[ 0.003032] Initializing cgroup subsys devices
[ 0.003091] Initializing cgroup subsys freezer
[ 0.003146] Initializing cgroup subsys net_cls
[ 0.003254] CPU: Testing write buffer coherency: ok
[ 0.003368] ftrace: allocating 19733 entries in 58 pages
[ 0.105896] Setting up static identity map for 0x81c0 - 0x81f8
[ 0.108018] devtmpfs: initialized
[ 0.118752] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[ 0.119214] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[ 0.121144] pinctrl core: initialized pinctrl subsystem
[ 0.122054] NET: Registered protocol family 16
[ 0.127646] DMA: preallocated 4096 KiB pool for atomic coherent allocations
[ 0.129197] bcm2708.uart_clock = 3000000
[ 0.134250] hw-breakpoint: found 6 breakpoint and 1 watchpoint registers.
[ 0.134325] hw-breakpoint: maximum watchpoint size is 4 bytes.
[ 0.134532] Serial: AMBA PL011 UART driver
[ 0.134837] 20201000.uart: ttyAMA0 at MMIO 0x20201000 (irq = 83, base_baud = 0) is a PL011 rev2
[ 0.511591] console [ttyAMA0] enabled
[ 0.515978] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[ 0.561559] bcm2708-dmaengine 20007000.dma: DMA legacy API manager at f2007000, dmachans=0xf35
[... 204 lines cut ...]
[ 12.853527] cfg80211: World regulatory domain updated:
[ 12.853538] cfg80211: DFS Master region: unset
[ 12.853547] cfg80211: (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
[ 12.853563] cfg80211: (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[ 12.853576] cfg80211: (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
[ 12.853603] cfg80211: (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[ 12.853616] cfg80211: (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[ 12.853629] cfg80211: (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[ 12.853640] cfg80211: (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[ 12.853651] cfg80211: (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[ 12.853661] cfg80211: (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
[... cut ...]
[ 15.067599] cfg80211: (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[ 15.067611] cfg80211: (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[ 15.067623] cfg80211: (5490000 KHz - 5710000 KHz @ 160000 KHz), (N/A, 2700 mBm), (0 s)
[ 15.067648] cfg80211: (57000000 KHz - 66000000 KHz @ 2160000 KHz), (N/A, 4000 mBm), (N/A)
[ 18.165362] Adding 102396k swap on /var/swap. Priority:-1 extents:2 across:839676k SSFS
[ 34.701889] EXT4-fs (mmcblk0p5): mounted filesystem with ordered data mode. Opts: (null)
```

XECDesign, 2016-02-11:
Strange... doesn't look like it's trying to load the module. Could you install `i2c-tools` and provide the output of `i2cdetect -y 1` and lsmod?

denizozger, 2016-02-11:
Output is below, hope this helps:

```
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberrypi:~ $ lsmod
Module                  Size  Used by
rpisense_core           3307  0
brcmfmac              192661  0
brcmutil                8845  1 brcmfmac
cfg80211              477515  1 brcmfmac
rfkill                 21439  2 cfg80211
i2c_bcm2708             6068  0
bcm2835_rng             2215  0
bcm2835_gpiomem         3703  0
snd_bcm2835            22317  3
snd_pcm                92397  1 snd_bcm2835
snd_timer              21932  1 snd_pcm
snd                    66988  9 snd_bcm2835,snd_timer,snd_pcm
uio_pdrv_genirq         3550  0
uio                    10002  1 uio_pdrv_genirq
i2c_dev                 6406  0
fuse                   85816  3
ipv6                  354610  28
```

XECDesign, 2016-02-11:
That looks like it's not attached correctly or something is fried. If you can confirm other i2c devices work, you should get a replacement.

denizozger, 2016-02-11:
Thanks a lot @XECDesign, I'll ask for a replacement

denizozger, 2016-02-19:
I'm not sure what fixed it but: I attached the SenseHAT in a different way (didn't/don't know it can be wrong) and updated the firmware as described here: https://www.reddit.com/r/raspberry_pi/comments/3yg0rz/sense_hat_and_jessie/cyvvm8g, and now it works :)

AlainGeenrits, 2016-02-25:
I got the eeprom data, but when I try to flash, I get following error. Any
idea ? I thought rpe-update was not used anymore.

sudo ./eepflash.sh -f=sense.eep -t=24c32 -w
This will disable the camera so you will need to REBOOT after this process
completes.
This will attempt to write to i2c address 0x50. Make sure there is an
eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know
what you are doing.
Do you wish to continue? (yes/no): yes
modprobe: ERROR: ../libkmod/libkmod.c:557 kmod_search_moddep() could not
open moddep file '/lib/modules/4.1.13-v7+/modules.dep.bin'
Modprobe of i2c-bcm2708 failed. Do an rpi-update.

XECDesign, 2016-03-01:
@AlainGeenrits Apologies for the very late reply. Haven't seen this happen, but perhaps running `sudo depmod` will work? Seems like there's a problem with your `/lib/modules/4.1.13-v7+/modules.dep.bin`

AlainGeenrits, 2016-03-01:
Hi Serge,

No problem, I am glad with your help. I am a bit in the dark here... Let me know if I have too much questions and you cannot support me here.

I want to make sure there is nothing wrong with the hat. In two weeks I will take it with me to the polar circle in Lapland on an aurora hunt. I want to test if I can detect changes in the magnetic field indicating aurora activity...

I now get a timeout on the write every time... Ran all recent upgrades...

sudo ./eepflash.sh -f=sense.eep -t=24c32 -w
This will disable the camera so you will need to REBOOT after this process completes.
This will attempt to write to i2c address 0x50. Make sure there is an eeprom at this address.
This script comes with ABSOLUTELY no warranty. Continue only if you know what you are doing.
Do you wish to continue? (yes/no): yes
Writing...
dd: writing to ‘/sys/class/i2c-adapter/i2c-0/0-0050/eeprom’: Connection timed out
1+0 records in
0+0 records out
0 bytes (0 B) copied, 0.0609188 s, 0.0 kB/s
Error doing I/O operation.

Alain

XECDesign, 2016-03-03:
@AlainGeenrits Hmm, I haven't seen that one before. Might be possible that it wasn't programmed properly at the factory because there's a hardware fault and that slipped QC. Hard to tell from here. If I had the board here, I'd start looking at the I2C signals and seeing what's going on, then trying to talk to the chip myself.

The depmod issue earlier was most likely cause by upgrading the kernel and not rebooting. That would result in that exact error.

The aurora hunt trip sounds great, best of luck!

asamite, 2016-03-13:
I was having the same issue with it not finding the hat.  When I booted all the LEDs lit up and stayed lit.  I finally unplugged and reseated the hat straighter on the GPIOs (this is there using standoffs would have been best) and then I booted back up.  Once that was done the LEDs came on for a moment and then went back off.  This time my code worked fine.  I don't know about your particular situation, but it might be as easy as that.  Mine was on fully but just angled a slight bit and that was Enough.

denizozger, 2016-03-13:
My experience is similar to @asamite too, I'm pretty much certain that the lack of connectivity was the cause of this issue. It's quite sensitive and hard to make sure contact is maintained.

soulfly, 2016-04-11:
Had the same issue

update **raspi-config** and enable **i2c** did the trick

bennuttall, 2016-06-06:
Closed with #6

closed 2016-06-06 by bennuttall

faridskyman, 2017-01-27:
@asamite comment really helps, thanks
that was my issue, seems hat rather sensitive, got to press it in and ensure all point are correctly attached.
after a few re-tries, it started to work, i could finally run python program... thx

jd-shah, 2019-04-08:
> 
> 
> Try adding dtoverlay=rpi-sense to the bottom of /boot/config.txt
> That worked for me.

Yes this worked for me too. 
Just for newbies like me, reboot after this to see the results.

waheedi, 2020-06-25:
> I'm not sure what fixed it but: I attached the SenseHAT in a different way (didn't/don't know it can be wrong) and updated the firmware as described here: https://www.reddit.com/r/raspberry_pi/comments/3yg0rz/sense_hat_and_jessie/cyvvm8g, and now it works :)

I had the same problem i guess, just flipped the way i connected it :D