astro-pi/python-sense-hat #29: Orientation kalman filter gives incorrect pitch axis values

davidhoness, 2016-01-04:
Hi all
I think this is actually a bug with the RTIMU library. The `get_orientation` functions do not provide 360 degrees of coverage for the *pitch* axis. However it's fine for *yaw* and *roll*.

The issue is complicated by [this line](https://github.com/RPi-Distro/python-sense-hat/blob/master/sense_hat/sense_hat.py#L730) in the Sense HAT python library which is used to update values that have been converted from radians to degrees (in a scale of -180 to +180) to a more friendly scale of 0 to 360.

However the issue remains if you just use `get_orientation_radians` which leads me to believe the issue lies within the RTIMU library.

**How to reproduce**
Run the following code and then manually rotate the Sense HAT through 360 degrees in the pitch axis. Change `pitch` to `roll` or `yaw` to see how these work correctly when you manually rotate the Sense HAT through 360 degrees in those axes.

```python
import math
from sense_hat import SenseHat

while True:
    orientation = sense.get_orientation_radians()
    val = orientation["pitch"]
    print(math.degrees(val))
```

r-b-us, 2016-01-11:
The limited range for the pitch axis is expected. I have updated my IMU stuff blog entry (https://richardstechnotes.wordpress.com/imu-stuff/) to address this. I hope that this is helpful.

bennuttall, 2016-06-06:
Thanks for the info!

closed 2016-06-06 by bennuttall

waywaaard, 2017-02-10:
@r-b-us Hi I am currently working on the webbased sensehat emulator and was wondering if I can get access to your blog post about the imu-stuff and the pitch values.

I've requested access to your blog on wordpress with the same name as my github account.