astro-pi/python-sense-hat #30: OSError: Cannot detect RPi-Sense FB device

mesiapk, 2016-01-24:
I know this has already been brought up before, but none of the other problems seem to provide a solution to me.

I am using a bananapro with Raspian distro.

When running a a test script with sense_hat, an OSError is raised that reads
"Cannot detect RPi-Sense FB device"

When running: sudo RTIMULibDrive11

This is what is returned.

    Settings file RTIMULib.ini loaded
    No IMU detected
    Using fusion algorithm RTQF
    No IMU found

The sense_hat is clearly not being detected, and I have no idea why.

uname -a
   
    Linux cubesat 3.4.103 #4 SMP PREEMPT Thu Dec 18 12:55:58 CST 2014 armv7l GNU/Linux
vcgencmd version

    -bash: vcgencmd: command not found
sudo dtc -I fs /proc/device-tree

    FATAL ERROR: Couldn't opendir() "/proc/device-tree": No such file or directory

The device tree is enabled in raspi-config.

boot/config.txt

    dtparam=spi=on
    dtparam=i2c_arm=on
    #device_tree=

also tried

    dtparam=spi=on
    dtparam=i2c_arm=on
    device_tree=

still returns

    FATAL ERROR: Couldn't opendir() "/proc/device-tree": No such file or directory

XECDesign, 2016-01-25:
We only support raspberry pi.

closed 2016-01-25 by XECDesign

mesiapk, 2016-01-25:
Does the sense_hat work on bananapi at all? Do you know if anyone has had any luck with it?

I have a raspi and I still have the same problem.

bennuttall, 2016-01-25:
No idea if it works on Banana Pi. Try asking the developers.

If you have the same issue on Raspberry Pi, please post your information here.

mesiapk, 2016-01-26:
How would I go about speaking with the developers?

XECDesign, 2016-01-26:
If you take a raspberry pi and install either raspbian lite and 'apt-get install sense-hat' and reboot or if you just use the full image and provide the same information you did in the first post, I would be happy to help.

bennuttall, 2016-01-26:
Yes, I meant the developers of Banana Pi. We're an educational charity, we don't have time to give customer support to users of other people's hardware.

As @XECDesign said, take an image from [raspberrypi.org/downloads](https://www.raspberrypi.org/downloads/) and if the Sense HAT still doesn't work, try posting the same information you provided earlier here.

blackjack75, 2017-04-22:
@KnowIsPow Did you have any success with it ?