astro-pi/python-sense-hat #31: _unpack_bin(_pack_bin([r,g,b])) != [r,g,b]

aonghusonia, 2016-01-24:
I just set some pixels on the sense hat & got them back & the rgb values were off by a few. I didn't know what  was going on so I  did a little test on the pack_bin & unpack_bin functions:

self._unpack_bin(self._pack_bin( [149, 165, 8] )) ->  [144, 164, 8]
self._unpack_bin(self._pack_bin( [138, 113, 42] )) ->  [136, 112, 40]
self._unpack_bin(self._pack_bin( [183, 67, 236] )) ->  [176, 64, 232]
self._unpack_bin(self._pack_bin( [68, 9, 47] )) ->  [64, 8, 40]
self._unpack_bin(self._pack_bin( [113, 162, 60] )) ->  [112, 160, 56]
self._unpack_bin(self._pack_bin( [76, 217, 12] )) ->  [72, 216, 8]
self._unpack_bin(self._pack_bin( [95, 161, 251] )) ->  [88, 160, 248]
self._unpack_bin(self._pack_bin( [179, 16, 163] )) ->  [176, 16, 160]
self._unpack_bin(self._pack_bin( [123, 210, 75] )) ->  [120, 208, 72]
self._unpack_bin(self._pack_bin( [96, 143, 2] )) ->  [96, 140, 0]
self._unpack_bin(self._pack_bin( [66, 61, 177] )) ->  [64, 60, 176]
self._unpack_bin(self._pack_bin( [129, 9, 106] )) ->  [128, 8, 104]
self._unpack_bin(self._pack_bin( [70, 173, 143] )) ->  [64, 172, 136]
self._unpack_bin(self._pack_bin( [53, 225, 10] )) ->  [48, 224, 8]
self._unpack_bin(self._pack_bin( [112, 37, 53] )) ->  [112, 36, 48]
self._unpack_bin(self._pack_bin( [196, 71, 65] )) ->  [192, 68, 64]
self._unpack_bin(self._pack_bin( [9, 87, 49] )) ->  [8, 84, 48]
self._unpack_bin(self._pack_bin( [204, 153, 178] )) ->  [200, 152, 176]
self._unpack_bin(self._pack_bin( [36, 58, 4] )) ->  [32, 56, 0]
self._unpack_bin(self._pack_bin( [220, 86, 136] )) ->  [216, 84, 136]

XECDesign, 2016-01-25:
The actual format used by the  board is RGB565, so some bits do get lost in the conversion. For example, 149 = 0b10010101, shift it to make it a 5 bit value and you get 0b10010, shift it back and you get 0b10010000 = 144, as you see in your own experiment.

aonghusonia, 2016-01-25:
Ok I think I understand your loosing the last 3 bits when you shift so that's 2^3 so when you shift so that rounds down to the nearest multiple of 8 for red & blue or 4 in the case of green.

closed 2016-01-25 by aonghusonia