astro-pi/python-sense-hat #34: Humidity init fails

ichy-wayland, 2016-01-26:
Hello. Im trying to write a Sense Hat program that uses temperature scans, but i get this: "Humidity init failed, please run as root / use sudo". Im running as root. How to proceed?

XECDesign, 2016-01-26:
Is i2c enabled and is the i2c dev node present?

ichy-wayland, 2016-01-26:
How can i know? I have i2c-tools installed, just know that. Is there any script?

XECDesign, 2016-01-26:
You'll need to find archlinux resources on your own. Here's an example:
https://wiki.archlinux.org/index.php/Raspberry_Pi#I2C

We only have the resources to support raspbian.

ichy-wayland, 2016-01-26:
Solved.

closed 2016-01-26 by ichy-wayland

jwhendy, 2017-09-24:
Breadcrumbs: for others perplexed by this, I think that the message no longer contains the bit about running as root. Thus, I thought I didn't have the right driver loaded, but it turns out it still *is* a permissions issue.

```
$ ls -l /dev/i2c-1 
crw------- 1 root root 89, 1 Aug 16 12:20 /dev/i2c-1
```

Some use a script to `chmod` the device; I implemented a `udev` rule:
```
$ cat /etc/udev/rules.d/99-i2c.rules 
KERNEL=="i2c-[0-7]",MODE="0666"

$ ls -l /dev/i2c-1 
crw-rw-rw- 1 root root 89, 1 Aug 16 12:20 /dev/i2c-1
```

I can now use the sensors on the hat. Previously none worked ( #79 )!