astro-pi/python-sense-hat #37: Full Python Sense Hat API availability

Notnumbersix, 2016-02-23:
Hi - sorry of this is fairly basic but I'm new to the Raspberry Pi, Sense Hat and trying to learn Python at the same time I'm enjoying it but there are frustrations!

I've got the Sense Hat working fine on a Pi Zero. I've managed to program the examples int he API document - http://pythonhosted.org/sense-hat/ 

My problem is finding the full API documentation, e.g. the MagPi has an example using -

 sense.get_orientation().values()

this is not in the above API documentation - there must be lots more too. How do I find a full list of methods, etc?

Thanks for any advice.

bennuttall, 2016-02-23:
I assume you found the documentation for `get_orientation` on the [API page](http://pythonhosted.org/sense-hat/api/) on pythonhosted?

The `get_orientation` method returns a [dictionary](https://docs.python.org/3/tutorial/datastructures.html), which is a built-in Python data type. A dictionary object has a `values` method which returns a list of its values (as opposed to its keys). This isn't documented as it's part of the Python standard library.

The documentation covers all the methods and properties of the `SenseHat` class. For further information on how to manipulate the return values, try looking up the return value type:

```python
>>> o = sense.get_orientation().values()
>>> type(o)
dict
```

then looking up that data type in the [Python docs](https://docs.python.org/3/).

Or look up the docstring from the shell with the `help` command:

```python
>>> o = sense.get_orientation().values()
>>> help(o)
Help on dict object:

class dict(object)
 |  dict() -> new empty dictionary
...
```

Hope that helps.

Notnumbersix, 2016-02-24:
Hi - yes, thanks for that. I suppose I was expecting the API page on pythonhosted to have in the list sense.get_orientation().values() as well as just sense.get_orientation() plus other options as well. I see now that I need to look further into the syntax for Python to understand about methods that are available as standard on objects produced by the API code. 

Thanks again, it's shown I have found the info but need to understand it better.

The full procedure I picked out from MagPi Essentials Edition on SenseHat is this - 

from sense_hat import SenseHat
sense = SenseHat()
while True:
      pitch, roll, yaw = sense.get_orientation().values()
      print("pitch=%s, roll=%s, yaw=%s" % (pitch,yaw,roll))

Hope I'm not taking too much of your time - does it not matter here that the 3 data labels are a different way round in the last bracket? The first two occurrences have "pitch, roll, yaw" but the last one has "pitch, yaw, roll" - is the print() not matching the %s values in sequence?

closed 2016-06-06 by bennuttall