astro-pi/python-sense-hat #38: OSError: Cannot detect Rpi-Sense FB device

shamde, 2016-03-01:
Hi I'm getting the following error when i try any examples (like ranbow.py) from the examples for sense-hat ---->>>> OSError: Cannot detect Rpi-Sense FB device

I'm using Raspberry pi 2.  at poweron of the pi all the LED light up then shuts off.
"sudo RTIMULibDrive11" seems to be working.

from reading the forum's i've done all below:
- enabled i2c and device tree in raspi-config
- set "dtoverlay=rpi-sense" and "dtdebug=1" to the bottom of /boot/config.txt
- sudo rpi-update
- sudo apt-get update
- sudo apt-get install sense-hat
- sudo pip-3.2 install pillow

Still no luck!  any ideas?

see results of
uname -a
vcgencmd version
sudo dtc -I fs /proc/device-tree

pi@Hayabusa:~/Scripts/SenseHat/examples/python-sense-hat $ **uname -a
**Linux Hayabusa 4.1.18-v7+ #848 SMP Mon Feb 29 14:19:45 GMT 2016 armv7l GNU/Linux

pi@Hayabusa:~/Scripts/SenseHat/examples/python-sense-hat $ **vcgencmd version
**Feb 29 2016 14:23:50 
Copyright (c) 2012 Broadcom
version dea971b793dd6cf89133ede5a8362eb77e4f4ade (clean) (release)
pi@Hayabusa:~/Scripts/SenseHat/examples/python-sense-hat $ **sudo dtc -I fs /proc/device-tree >dtc1.txt**

pi@Hayabusa:~/Scripts/SenseHat/examples/python-sense-hat $ **sudo RTIMULibDrive11**
Settings file not found. Using defaults and creating settings file
Detected LSM9DS1 at standard/standard address
Using fusion algorithm RTQF
Detected LPS25H at standard address
Detected HTS221 at standard address
min/max compass calibration not in use
Ellipsoid compass calibration not in use
Accel calibration not in use
LSM9DS1 init complete
Sample rate 0: : roll:-2.048982, pitch:-1.241802, yaw:-95.723995
Pressure: 997.7, height above sea level: 130.1, temperature: 32.9, humidity: 14.1
Sample rate 0: : roll:-1.992474, pitch:-1.318911, yaw:-98.709450
Pressure: 997.7, height above sea level: 130.1, temperature: 32.9, humidity: 14.1

results of 
**sudo dtc -I fs /proc/device-tree >dtc1.txt**

/dts-v1/;

/ {
	model = "Raspberry Pi 2 Model B Rev 1.1";
	compatible = "brcm,bcm2709";
	memreserve = <0x3b000000 0x4000000>;
	#address-cells = <0x1>;
	#size-cells = <0x1>;
	interrupt-parent = <0x1>;

	axi {

		dma {
			broadcom,channels = <0x7f35>;
		};

		usb {

			hub {

				ethernet {
					mac-address = [b8 27 eb 50 12 80];
				};
[... 652 lines cut ...]
		i2c2_iknowwhatimdoing = [00 00 00 1c 73 74 61 74 75 73 00];
		act_led_activelow = <0x1d 0x6770696f 0x733a3800>;
		pwr_led_gpio = <0x1e 0x6770696f 0x733a3400>;
		i2c0_baudrate = [00 00 00 1a 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		sd_overclock = "", "", "", "\"brcm,overclock-50:0";
		pwr_led_trigger = [00 00 00 1e 6c 69 6e 75 78 2c 64 65 66 61 75 6c 74 2d 74 72 69 67 67 65 72 00];
		sd_pio_limit = "", "", "", "\"brcm,pio-limit:0";
		sd_debug = "", "", "", "\"brcm,debug";
		i2c_vc_baudrate = [00 00 00 1a 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		uart0_clkrate = [00 00 00 03 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		i2c_vc = [00 00 00 1a 73 74 61 74 75 73 00];
		i2c1_baudrate = [00 00 00 1b 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		i2c_baudrate = [00 00 00 1b 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		watchdog = "", "", "", " status";
		sd_force_pio = "", "", "", "\"brcm,force-pio?";
		random = "", "", "", "!status";
		arm_freq = <0x13 0x636c6f63 0x6b2d6672 0x65717565 0x6e63793a 0x30000000 0x14636c 0x6f636b2d 0x66726571 0x75656e63 0x793a3000 0x15 0x636c6f63 0x6b2d6672 0x65717565 0x6e63793a 0x30000000 0x16636c 0x6f636b2d 0x66726571 0x75656e63 0x793a3000>;
		i2c2_baudrate = [00 00 00 1c 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		core_freq = [00 00 00 05 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		i2c_arm_baudrate = [00 00 00 1b 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		pwr_led_activelow = <0x1e 0x6770696f 0x733a3800>;
		act_led_gpio = <0x1d 0x6770696f 0x733a3400>;
		cache_line_size = [00 00 00 12 63 61 63 68 65 2d 6c 69 6e 65 2d 73 69 7a 65 3a 30 00];
	};
};

XECDesign, 2016-03-03:
Could you also post the output of dmesg, cat /sys/class/graphics/*/name and ls /dev/fb* ?

shamde, 2016-03-04:
Thanks for looking into it!
Here is the requested output of dmesg, cat /sys/class/graphics//name and ls /dev/fb

pi@Hayabusa:~/Scripts/SenseHat $ **ls /sys/class/graphics/***
/sys/class/graphics/fb1

/sys/class/graphics/fb0:
bits_per_pixel  bl_curve  cursor  device  modes  pan     state   subsystem  virtual_size
blank           console   dev     mode    name   rotate  stride  uevent

/sys/class/graphics/fbcon:
cursor_blink  rotate  rotate_all  subsystem  uevent
pi@Hayabusa:~/Scripts/SenseHat $ **cat /sys/class/graphics/fb0/name** 
BCM2708 FB
pi@Hayabusa:~/Scripts/SenseHat $ **ls /dev/fb***
/dev/fb0  /dev/fb1

**content of dmesg below:**
[    0.000000] Booting Linux on physical CPU 0xf00
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.18-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #848 SMP Mon Feb 29 14:19:45 GMT 2016
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 2 Model B Rev 1.1
[    0.000000] cma: Reserved 8 MiB at 0x3a400000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80876f80, node_mem_map b9bb4000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9420->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 12 pages/cpu @bafb0000 s20416 r8192 d20544 u49152
[    0.000000] pcpu-alloc: s20416 r8192 d20544 u49152 alloc=12*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=1024 bcm2709.boardrev=0xa21041 bcm2709.serial=0x5e501280 smsc95xx.macaddr=B8:27:EB:50:12:80 bcm2708_fb.fbswap=1 bcm2709.uart_clock=3000000 bcm2709.disk_led_gpio=47 bcm2709.disk_led_active_low=0 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939416K/966656K available (6024K kernel code, 534K rwdata, 1660K rodata, 448K init, 757K bss, 19048K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff000000   (1080 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807895a0   (7686 kB)
      .init : 0x8078a000 - 0x807fa000   ( 448 kB)
      .data : 0x807fa000 - 0x8087fac0   ( 535 kB)
       .bss : 0x80882000 - 0x8093f79c   ( 758 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] NR_IRQS:608
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000012] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000033] Switching to timer-based delay loop, resolution 52ns
[    0.000317] Console: colour dummy device 80x30
[    0.001528] console [tty1] enabled
[    0.001587] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001664] pid_max: default: 32768 minimum: 301
[    0.002050] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002104] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.003395] Initializing cgroup subsys blkio
[    0.003471] Initializing cgroup subsys memory
[... 350 lines cut ...]
[   37.678299] [<8009a458>] (SyS_finit_module) from [<8000fa20>] (ret_fast_syscall+0x0/0x54)
[   37.678310] ---[ end trace fd2df6b5699042aa ]---
[   37.678390] rpi-sense 1-0046: Failed to register rpi-sense-fb: -17
[   37.678446] bcm2708_i2c 3f804000.i2c: BSC1 Controller at 0x3f804000 (irq 79) (baudrate 400000)
[  123.212219] wlan0: authenticate with 6e:7c:5f:2f:87:8e
[  123.272965] wlan0: send auth to 6e:7c:5f:2f:87:8e (try 1/3)
[  123.317496] wlan0: authenticated
[  123.324911] wlan0: associate with 6e:7c:5f:2f:87:8e (try 1/3)
[  123.329189] wlan0: RX AssocResp from 6e:7c:5f:2f:87:8e (capab=0x411 status=0 aid=1)
[  123.336511] wlan0: associated
[  123.336668] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  123.336934] cfg80211: Calling CRDA for country: XZ
[  123.387999] wlan0: Limiting TX power to 20 (20 - 0) dBm as advertised by 6e:7c:5f:2f:87:8e
[  126.484909] cfg80211: Calling CRDA to update world regulatory domain
[  126.504607] cfg80211: World regulatory domain updated:
[... cut ...]
[  126.504704] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
[  126.504725] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
[  126.504744] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
[  126.504760] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
[  126.504775] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
[  126.504791] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)

XECDesign, 2016-03-04:
Do you have anything in /etc/modules ?

shamde, 2016-03-04:
pi@Hayabusa:~ $ more /etc/modules
# /etc/modules: kernel modules to load at boot time.
#
# This file contains the names of kernel modules that should be loaded
# at boot time, one per line. Lines beginning with "#" are ignored.
i2c-bcm2708
i2c-dev

XECDesign, 2016-03-04:
Thanks.

We see `[ 5.465913] rpi-sense 1-0046: Raspberry Pi Sense HAT firmware version 0` which is expected and tells you that the module is being loaded. Then there is `[ 6.277322] fb1: RPi-Sense FB frame buffer device` which suggests that the frame buffer device loads successfully. That around the time the LEDs would be going off. Then a full 30 seconds later, `rpisense_probe` functions runs again and we see this:

```
[ 37.676452] rpi-sense 1-0046: Raspberry Pi Sense HAT firmware version 0
[ 37.676963] input: Raspberry Pi Sense HAT Joystick as /devices/virtual/input/input4
[ 37.677283] ------------[ cut here ]------------
[ 37.677312] WARNING: CPU: 1 PID: 846 at fs/sysfs/dir.c:31 sysfs_warn_dup+0x74/0x84()
[ 37.677322] sysfs: cannot create duplicate filename '/bus/platform/devices/rpi-sense-fb'`
```

And that seems to be breaking things by trying to register the framebuffer twice. No idea why this is happening off the top of my head. I need to have a think about why it's running the `rpisense_probe` function multiple times. Since it's 30 seconds later, that would indicate that it's done by something in userland. Are there any modifications you've made that aren't in the original post? If I can replicate the error on Monday, that would make it easier to find the problem.

shamde, 2016-03-05:
Thanks for your input.  it's working now.  but i still have another problem.

A startup script was changing the baud rate of the i2c to 400000 at the 37th mark.  Which is forcing the sensehat to register the frame buffer 2nd time...... and failing.  i removed the startup script and now it's working.

Now the question is, does the sense hat operate at 400000 i2c baud rate?  if it does, is there a recommended way of changing the baud rate to 400000?  i need that baud rate for another i2c interface.

Thanks!

XECDesign, 2016-03-05:
Could try this approach https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README#L98

Don't know how well the HAT will work at that rate.

closed 2016-03-08 by XECDesign