astro-pi/python-sense-hat #39: get_compass() always returns 0

camfrout, 2016-03-05:
Hi,

using the code below, I see that the Sense Hat on my RPi2 is continuously returning 0.
My Pi is running OSMC, a kodi distribution based on Debian Jessie. 

```
from sense_hat import SenseHat

sense = SenseHat()
sense.set_rotation(0)
sense.clear()

while True:
    dir = sense.get_compass()
    print(dir)
```

I installed the packages `python-sense-hat` and `python3-sense-hat` but not `sense-hat` as according to this thread it will mess with OSMC: https://discourse.osmc.tv/t/sense-hat-how-to/8929/5

I tried to run some scripts to use the LEDs of the Sense Hat and it is working.

BenIanGifford, 2016-03-07:
when i ran that code on raspbian it worked. annd returned numbers consistently correct.

JGifford, 2016-03-08:
I've had some troubles with my sense hat's magnetometer, but I did get consistent results after I ran the [calibration steps](https://www.raspberrypi.org/forums/viewtopic.php?f=104&t=109064&p=750616#p810193).

camfrout, 2016-03-08:
I installed the packet librtimulib-utils to get the program RTIMULibCal but at the end of the installation, the folder RTEllipsoidFit contains only the following files: 
```
ellipsoid_fit.m  
mag_cal.m  
mag_fit_display.m  
RTEllipsoidFit.m
```

So I downloaded the files from here https://github.com/RPi-Distro/RTIMULib and compiled RTIMULibCal.
But then a new error appears: 
```
$:~/xxx/RTIMULib/Linux/RTIMULibCal/Output$ sudo ./RTIMULibCal
RTIMULibCal - using RTIMULib.ini
Settings file RTIMULib.ini loaded
Failed to open I2C bus 1
Failed to open SPI bus 0, select 0
Failed to open SPI bus 0, select 1
No IMU detected
Using fusion algorithm RTQF
No IMU found
```

If it is useful, the result of sudo `i2cdetect -y 1` is the following ```Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory```

camfrout, 2016-03-08:
After a bit of troubleshooting I found where the issue comes from. I had to add the value `i2c-dev`to my `/etc/modules` file and restart the RPi.
Now I just need to calibrate the sensor but at least it is returning some values. Thanks for the help, I am closing the issue.

closed 2016-03-08 by camfrout