astro-pi/python-sense-hat #43: OSError: Cannot detect RPi-Sense FB device ,

Incarnation, 2016-04-04:
Hello would you please have a look at my issues: I have tried the following. Thanks.

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.1.20-v7+ #867 SMP Wed Mar 23 20:12:32 GMT 2016 armv7l GNU/Linux

pi@raspberrypi:~ $ vcgencmd version
Mar 24 2016 12:37:55 
Copyright (c) 2012 Broadcom
version e16cd015026a8bda22bb50dc30b30ce07f52a202 (clean) (release)

pi@raspberrypi:~ $ sudo dtc -I fs /proc/device-tree
Warning (avoid_default_addr_size): Relying on default #address-cells value for /axi/vc_mem
Warning (avoid_default_addr_size): Relying on default #size-cells value for /axi/vc_mem
/dts-v1/;

/ {
	model = "Raspberry Pi 3 Model B Rev 1.2";
	compatible = "brcm,bcm2710", "brcm,bcm2709";
	memreserve = <0x3b000000 0x4000000>;
	#address-cells = <0x1>;
	#size-cells = <0x1>;
	interrupt-parent = <0x1>;

	axi {

		dma {
			broadcom,channels = <0x7f35>;
		};

		usb {

			hub {

				ethernet {
					mac-address = [b8 27 eb 89 69 0f];
				};
			};
		};

		vc_mem {
			reg = <0x3dc00000 0x3f000000 0xc0000000>;
		};
	};

	soc {
		compatible = "simple-bus";
		ranges = <0x7e000000 0x3f000000 0x1000000>;
		#address-cells = <0x1>;
		phandle = <0x28>;
		#size-cells = <0x1>;
		linux,phandle = <0x28>;

		fb {
			compatible = "brcm,bcm2708-fb";
			firmware = <0x15>;
			status = "okay";
			phandle = <0x30>;
			linux,phandle = <0x30>;
		};

		mailbox@7e00b800 {
			reg = <0x7e00b880 0x40>;
			interrupts = <0x0 0x1>;
			compatible = "brcm,bcm2835-mbox";
			#mbox-cells = <0x0>;
			phandle = <0x13>;
			linux,phandle = <0x13>;
		};

		leds {
[... 623 lines cut ...]
		audio = "", "", "", "#status";
		uart0 = [00 00 00 1b 73 74 61 74 75 73 00];
		uart1 = [00 00 00 1c 73 74 61 74 75 73 00];
		i2c2_iknowwhatimdoing = "", "", "", "!status";
		act_led_activelow = "", "", "", "\"gpios:8";
		i2c0_baudrate = [00 00 00 1f 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		sd_overclock = "", "", "", "&brcm,overclock-50:0";
		sd_pio_limit = "", "", "", "&brcm,pio-limit:0";
		sd_debug = "", "", "", "&brcm,debug";
		i2c_vc_baudrate = [00 00 00 1f 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		uart0_clkrate = [00 00 00 03 63 6c 6f 63 6b 2d 66 72 65 71 75 65 6e 63 79 3a 30 00];
		i2c_vc = [00 00 00 1f 73 74 61 74 75 73 00];
		i2c1_baudrate = "", "", "", " clock-frequency:0";
		i2c_baudrate = "", "", "", " clock-frequency:0";
		watchdog = "", "", "", "$status";
		sd_force_pio = "", "", "", "&brcm,force-pio?";
		random = "", "", "", "%status";
		arm_freq = <0x17 0x636c6f63 0x6b2d6672 0x65717565 0x6e63793a 0x30000000 0x18636c 0x6f636b2d 0x66726571 0x75656e63 0x793a3000 0x19 0x636c6f63 0x6b2d6672 0x65717565 0x6e63793a 0x30000000 0x1a636c 0x6f636b2d 0x66726571 0x75656e63 0x793a3000>;
		i2c2_baudrate = "", "", "", "!clock-frequency:0";
		core_freq = "", "", "", "\aclock-frequency:0";
		i2c_arm_baudrate = "", "", "", " clock-frequency:0";
		act_led_gpio = "", "", "", "\"gpios:4";
		cache_line_size = [00 00 00 16 63 61 63 68 65 2d 6c 69 6e 65 2d 73 69 7a 65 3a 30 00];
	};
};

XECDesign, 2016-04-05:
Does it work if you add `dtoverlay=rpi-sense` to `/boot/config.txt`?

Incarnation, 2016-04-05:
yes I added that and works! Thanks a lot. :)

XECDesign, 2016-04-05:
Np. In that case, this is a duplicate of #28.

closed 2016-04-05 by XECDesign