astro-pi/python-sense-hat #45: No suod apt-get install sense-hat

cristolmeth, 2016-05-07:
Hi I am new to RPI's and Ubuntu in general. I have an Ubuntu image on my RPI and im trying to apt-get install the proper packages for the Sense hat. Whenever I run sudo apt-get install sense-hat -y I get a Unable to locate package sense-hat error. I think its because Ubunutu doesn't come with that package and it won't install it. Are there any work arounds?

XECDesign, 2016-05-07:
Unfortunately, we don't maintain Ubuntu, so this isn't the place to report this issue.

closed 2016-05-07 by XECDesign

cristolmeth, 2016-05-07:
Do you know where the place to report this is I'm a little lost

>
> Unfortunately, we don't maintain Ubuntu, so this isn't the place to
report this issue.

XECDesign, 2016-05-07:
I am not sure, but I think I would start here. https://launchpad.net/ubuntu-mate

bennuttall, 2016-05-09:
Ping @flexiondotorg