astro-pi/python-sense-hat #47: Color names submodule?

eah13, 2016-05-13:
Managing RGB colors can be a bear.  I've been thinking about writing a module that would pre-define a bunch of variable names as their corresponding RGB tuples.  Is this something that you'd consider including as `sense_hat.colors`?  Then we'd be able to do something like this:

```python
from sense_hat import SenseHat
from sense_hat.colors import *

s = SenseHat()

screen = [ ]

# Note the below are names, not strings
for color in [pink, red, orange, yellow, green, blue, indigo, violet]:
  screen.append([color] * 8)

s.set_pixels(screen)
```

The color names could perhaps be taken from turtle, which in turn gets them from tkinter I believe.  one problem with the turtle names is that they have spaces and would need to be concatenated or snakecased.  So another option is to take them from CSS.  A reference is here: https://trinket.io/docs/colors

Also in the spirit of turtle we could be redundant, providing `midnight_blue`, `MidnightBlue` and `midnightblue`, all set equal to `(25,25,112)`.

The whole thing might run on a dict underneath, perhaps, and so you could `from sense_hat.colors import color_dict` as well.

Anyways, this is s a pain I felt when senseHatting for the first time and I had to write a bunch of boilerplate.  Figured it might be helpful to write it once and be done.

bennuttall, 2016-05-14:
This is definitely a good idea. The best colo(u)r library out there lies within @waveform80's [picamera](https://github.com/waveform80/picamera/). I think I persuaded him to bring it out into a separate module? Comes in handy for gpiozero, sense hat, picamera, pygame (zero), neopixels and more, on and off the Pi.

Having to declare a bunch of colour tuples at the top of is really sucky.

bennuttall, 2016-05-14:
See http://picamera.readthedocs.io/en/release-1.10/api_color.html

waveform80, 2016-05-16:
I should add that the forth-coming picamera 1.11 release will include constructors and attributes in the Color class for conversion to/from RGB565 which is the format used by the sense-HAT's pixel framebuffer (see http://picamera.readthedocs.io/en/latest/api_color.html), along with a load of CIE-Lab and Delta-E stuff.

eah13, 2016-05-16:
Wow nice.  Dipped my toe in the `Color` class and looks very powerful. Would make tons of sense if the tuples envisioned above were `Color` objects instead.

If the approach I proposed above RE color names is the way we go, I could see using something like [`NAMED_COLORS`](https://github.com/waveform80/picamera/blob/master/picamera/color.py#L54) to pre-instantiate a bunch of `Color`s in the module's namespace for users so that `green`, `goldenrod`, etc all work immediately upon import.  These would be `Color` objects underneath, giving them all that nice power, but beginning users wouldn't need to know that- their interface for using named colors would be very close to using natural language: `set_pixel(0, 0, green)`.  

@waveform80: did @bennuttall successfully persuade you to split that out as its own module? Would def be cleanest.

waveform80, 2016-05-16:
He has indeed persuaded me - the problem, as ever, is finding the time to do it. Splitting it out isn't terribly complicated but packaging is always fun to some degree. At the moment, I'm spending every spare moment trying to get picamera 1.11 out of the door (it's turned into a much more complex release than I'd planned but I think the performance gains will be worth it for the opencv crowd). After that's done I need to spend some time shoring up the bank balance again, then I can concentrate a bit on gpio-zero and splitting this out (I might just split this out first - it's probably no more than a day's work).

bennuttall, 2018-03-23:
Boom! http://colorzero.readthedocs.io

I'll close this issue now but we should look at the best ways of demonstrating use of colorzero with the Sense HAT.

@eah13 are you able to support this library in trinket?

closed 2018-03-23 by bennuttall