astro-pi/python-sense-hat #48: update installation instructions for jesse

seattleacademy, 2016-05-20:
Instructions for install python-sense-hat are not current for the latest release of jesse.

Rather than:
    sudo pip-3.2 install pillow

Instructions for jesse should read
    sudo apt-get install python3-pip
    sudo pip3 install Pillow

bennuttall, 2016-05-20:
The instructions do still work. `pip` and `pip3` are installed by default and there's an alias for `pip-3.2` -> `pip3`

closed 2016-05-20 by bennuttall

seattleacademy, 2016-05-20:
Perhaps the issue is that we are using jesse-lite (we don't use monitors or keyboards for robotic projects).  In that case, the packages that might be installed be default on the full version of jesse need to be explicitly installed leading to the errors we reported.

bennuttall, 2016-05-20:
Ah, I see. We'll update the instructions to Jessie instructions soon.

XECDesign, 2016-05-20:
For jessie, the instructions should really just be sudo apt-get install sense-hat, which will install everything that' required and set things up appropriately.