astro-pi/python-sense-hat #53: API reference for the SenseStick

martinohanlon, 2016-06-04:
Is the documentation for the stick missing ?

bennuttall, 2016-06-04:
Yes - @waveform80 wrote it and we merged his PR (#12) but haven't done a release yet. @davidhoness and I plan to work through the issues and PRs on Monday and hopefully update the docs and do a release!

waveform80, 2016-06-04:
Yeah, there's a load of stuff in github.com/waveform80/pisense that I need to get around to cleaning up and finishing off (docs, tests, packaging, etc.). Unfortunately, with other projects it's unlikely to happen any time soon (shame really - the screen stuff I built is pretty neat, and the orientation stuff is much easier to use than the degree based stuff - but it was all thrown together in a quick weekend hack which I've never found the time to tidy up).

bennuttall, 2016-07-13:
Hi @martinohanlon the Stick implementation is now complete, and in the new release. It's currently available via pip (and later the apt package will be updated). It's also documented at http://pythonhosted.org/sense-hat/api/

Please do test, play around, ask questions, create examples and give us feedback!

closed 2016-07-13 by bennuttall