astro-pi/python-sense-hat #57: osError: Cannot detect RPI-Sense FB device

lightningjim, 2016-10-06:
Yes, another one of these, but I've gone ahead and tried to narrow down the issue and I think I have.
Everything is working fine under RTIMULib. However I can't run any of the Python scripts because fo the aformentioned failure.

Checked dmesg and I found this:
[    3.826516] rpi-sense 1-0046: Read from reg 240 failed
[    3.826563] rpi-sense: probe of 1-0046 failed with error -22

Which also doesn't show here either
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: -- -- -- -- -- -- -- --

Do I possibly have a broken Sense Hat or is something else going on?

XECDesign, 2017-02-28:
If it doesn't "just work" and it has never been plugged in the wrong way or otherwise short circuited , I would recommend returning it.

mushu999, 2017-06-04:
Did you try this? Add
dtoverlay=rpi-sense
in /boot/config.txt to force sense HAT detection.
Might be required if you have anything else plugged in...see:
[ https://www.raspberrypi.org/forums/viewtopic.php?f=104&t=133244 ]

alt3red, 2017-08-05:
@mushu999 thanks, this helped me today.

closed 2017-09-25 by XECDesign