astro-pi/python-sense-hat #58: Sense Hat can't find RTIMU after upgrade to Python 3.5.2

DaddyTheRunner, 2016-10-23:
I recently upgraded the GNU R packages from 3.1.1 to 3.3.1. To do this, I had to add the stretch distribution to my /etc/apt/sources.list file. In the process of successfully upgrading GNU R, apt also chose to install an upgraded version of Python3. Now when I try to use the Sense Hat module with Python3, I get the following error message:

`Python 3.5.2+ (default, Sep 22 2016, 12:18:14) 
[GCC 6.1.1 20160802] on linux
Type "copyright", "credits" or "license()" for more information.
>>> from sense_hat import SenseHat
Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    from sense_hat import SenseHat
  File "/usr/lib/python3/dist-packages/sense_hat/__init__.py", line 2, in <module>
    from .sense_hat import SenseHat, SenseHat as AstroPi
  File "/usr/lib/python3/dist-packages/sense_hat/sense_hat.py", line 10, in <module>
    import RTIMU  # custom version
ImportError: No module named 'RTIMU'
>>>`

pip3 reports that sense-hat 2.2.0 and RTIMUlib 7.2.1 are installed.  I do NOT get this error when I use Python 2.7.9.

Any ideas on what got broken during the apt-get install of r-base?  It was working fine before the GNU R upgrade.

bennuttall, 2016-10-24:
Can you check where RTIMU is installed to? Maybe it's installed for Python 3.4? You might want to grab a copy of RPF's fork of RTIMU lib and install it yourself.

Download from https://github.com/RPi-Distro/RTIMULib
See installation instructions [here](https://github.com/RPi-Distro/RTIMULib/blob/b949681af69b45f0f7f4bb53b6770037b5b02178/Linux/python/README.md)

DaddyTheRunner, 2016-10-25:
Here are the results of my investigative work:

First, I ran IDLE using Python 2.7 and issued the following commands to locate the RTIMULib:

```
Python 2.7.9 (default, Sep 17 2016, 20:26:04) 
[GCC 4.9.2] on linux2
Type "copyright", "credits" or "license()" for more information.
>>> import RTIMU
>>> RTIMU.__file__
'/usr/lib/python2.7/dist-packages/RTIMU.arm-linux-gnueabihf.so'
>>> 
```

I then searched for all files starting with RTI in the /usr/lib sub-directories:
```
pi@raspberrypi:/usr/lib $ find . -name 'RTI*'
./python2.7/dist-packages/RTIMULib-7.2.1.egg-info
./python2.7/dist-packages/sense_emu/RTIMU.pyc
./python2.7/dist-packages/sense_emu/RTIMU.py
./python2.7/dist-packages/RTIMU.arm-linux-gnueabihf.so
./python3/dist-packages/RTIMULib-7.2.1.egg-info
./python3/dist-packages/RTIMU.cpython-34m-arm-linux-gnueabihf.so
./python3/dist-packages/sense_emu/__pycache__/RTIMU.cpython-34.pyc
./python3/dist-packages/sense_emu/RTIMU.py
```

I noticed that there is a similarly named, but not identical, .so file in the python3 directory.

Checking the installed python modules via help("modules") in IDLE3 doesn't show anything with RTIMU in the name.

Here are the results of checking the status of the python3-rtimulib package:
```
pi@raspberrypi:~/info $ dpkg -s python3-rtimulib
Package: python3-rtimulib
Status: install ok installed
Priority: optional
Section: python
Installed-Size: 192
Maintainer: Serge Schneider <serge@raspberrypi.org>
Architecture: armhf
Source: rtimulib
Version: 7.2.1-3
Depends: libc6 (>= 2.4), libgcc1 (>= 1:4.4.0), libstdc++6 (>= 4.3.0), librtimulib7 (= 7.2.1-3)
Recommends: librtimulib-utils
Description: Versatile C++ and Python 9-dof, 10-dof and 11-dof IMU library (Python 3)
 The simplest way to connect a 9-dof, 10-dof or 11-dof IMU to an embedded Linux
 system and obtain Kalman-filtered quaternion or Euler angle pose data.
 Basically, two simple funtion calls (IMUInit() and IMURead()) are pretty much
 all that's needed to integrate RTIMULib.
 .
 This package provides the Python 3 module.
Homepage: https://github.com/richards-tech/RTIMULib
```

I'm not sure what all this means other than Python 2 finds the library and Python 3 doesn't.

I am currently in the process of doing a manual install.  I'll post the results after it finishes.

DaddyTheRunner, 2016-10-25:
I am happy to report that the manual install was successful.  After the install, I decided to check to see what changed.  Other than now working, I couldn't find what changed.  The output of

```
pi@raspberrypi:/usr/lib $ find . -name 'RTI*'
```

reported the same list of files as before.  Python 3 now reported the location of RTIMU as:

```
/usr/local/lib/python3.5/dist-packages/RTIMU.cpython-35m-arm-linux-gnueabihf.so
```

which existed before the manual install but wasn't being found.

Thanks for helping me fix the problem.  I'm still confused as to why using apt-get to install the latest available version of GNU R broke the link to the RTIMU library.

XECDesign, 2016-10-25:
You have installed an unsupported version of Python. Our packages are built to work with the versions of python available in jessie. Adding the stretch repo is expected to break many things.

dborgesr, 2017-06-05:
Has anyone gotten this to work? I'm hitting this same issue when trying to dockerize sense-hat

closed 2017-09-25 by XECDesign

bennuttall, 2018-03-23:
I have uploaded a test copy of RTIMULib to PyPI in order to make this process simpler.

You can now `pip install rtimulib` in a virtualenv, and as long as you have the other requirements you stated, it works. However, I seemed to need `libatlas-base-dev`, `libopenjp2-7`, `libtiff-tools` and `sense-hat`.

Once this is confirmed and tested properly I'll push out a new sense hat version to pypi including the dependency on rtimulib, and document the process required for using in a virtualenv.

eplanet, 2018-03-23:
That's very nice of you! I just tested successfully with:
```bash
sudo apt install libopenjp2-7 sense-hat
pip install sense-hat rtimulib
```
That seem to be sufficient, `libtiff-tools` is installed as a `sense-hat` dependency and I didn't need `libatlas-base-dev` at all. Hope this helps!

lemonkey, 2018-04-18:
I'm getting "PyRTIMU.h: No such file or directory" with the above suggestions.

lemonkey, 2018-04-18:
I ended up fixing it by installing the RTIMU Raspi fork manually.

eplanet, 2018-04-19:
I'm surprised you needed to do that... Maybe I should try again with a fresh install to make sure.

Teteresa, 2019-03-20:
Hello :)
I followed the Instructions from the Git readme (RTIMULib). 
After a successful installation via `sudo apt-get install python3-dev` The next step `~/Downloads/RTIMULib-master/Linux/python$ python setup.py build` produced following error:
`running build
running build_ext
building 'RTIMU' extension
gcc -pthread -B /home/theresa/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -DHAL_QUIET -I../../RTIMULib -I/home/theresa/anaconda3/include/python3.7m -c PyRTIMU.cpp -o build/temp.linux-x86_64-3.7/PyRTIMU.o -std=c++0x
gcc: error trying to exec 'cc1plus': execvp: Datei oder Verzeichnis nicht gefunden
error: command 'gcc' failed with exit status 1
`
I am using Pycharm and Python3.7 and I am relatively new to the topic. Can someone help me what to do next?

postema, 2020-05-06:
same problem, with python3.8 on a PRi-2 in a virtual environment
the pip install RTIMULib return with an error

```
Running setup.py install for RTIMULib ... error
    ERROR: Command errored out with exit status 1:
     command: /var/www/lab_app/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jd0jlh8r/RTIMULib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jd0jlh8r/RTIMULib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-3l18gjw7/install-record.txt --single-version-externally-managed --compile --install-headers /var/www/lab_app/include/site/python3.8/RTIMULib
         cwd: /tmp/pip-install-jd0jlh8r/RTIMULib/
    Complete output (12 lines):
    running install
    running build
    running build_ext
    building 'RTIMU' extension
    creating build
    creating build/temp.linux-armv6l-3.8
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DHAL_QUIET -I../../RTIMULib -I/var/www/lab_app/include -I/usr/local/opt/python-3.8.2/include/python3.8 -c PyRTIMU.cpp -o build/temp.linux-armv6l-3.8/PyRTIMU.o -std=c++0x
    PyRTIMU.cpp:28:10: fatal error: PyRTIMU.h: Bestand of map bestaat niet
     #include "PyRTIMU.h"
              ^~~~~~~~~~~
    compilation terminated.
    error: command 'gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /var/www/lab_app/bin/python3.8 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-jd0jlh8r/RTIMULib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-jd0jlh8r/RTIMULib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-3l18gjw7/install-record.txt --single-version-externally-managed --compile --install-headers /var/www/lab_app/include/site/python3.8/RTIMULib Check the logs for full command output.

```

rzr, 2020-05-07:
There is no tracker on this project:
https://github.com/RPi-Distro/RTIMULib/blob/master/Linux/python/PyRTIMU.h

May a new bug be open to request support pip3 along Python 3.7.3 ? 

Related:

https://github.com/tanasegabriel/sense-api/pull/5

https://github.com/tanasegabriel/sense-api/pull/6

alvaroga91, 2020-08-21:
Having this issue when trying to Dockerize this on an Arm64 (aarch64-linux-gnu-gcc) device. 

Cloning the rtimulib repo and building it doesn't work for me, probably because I'm missing the gcc environment (which I want to avoid having due to the needed extra space...). Are people building this without GCC?

```
Step 7/12 : COPY rtimulib rtimulib
 ---> c334a143444a
Step 8/12 : RUN cd rtimulib && python3 setup.py build && python3 setup.py install
 ---> Running in f6a3b029b8ef
running build
running build_ext
building 'RTIMU' extension
creating build
creating build/temp.linux-aarch64-3.7
creating RTIMULib
creating RTIMULib/IMUDrivers
aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DHAL_QUIET -I../../RTIMULib -I/usr/include/python3.7m -c PyRTIMU.cpp -o build/temp.linux-aarch64-3.7/PyRTIMU.o -std=c++0x
error: command 'aarch64-linux-gnu-gcc' failed with exit status 1
```

rzr, 2021-03-03:
Maybe fixes in RTIMULib would help ?

I am checking if upstream is still active with:

https://github.com/mrbichel/RTIMULib/pull/2

If not then it could be a good candidate for @abandonware team see:

https://rzr.github.io/rzr-presentations//docs/abandonware/

RaynaRaven, 2022-12-18:
> As it's not very clear in that thread, I'd like to share what I did. Basically following [`bennuttall`](https://github.com/RPi-Distro/python-sense-hat/issues/58#issuecomment-255783155) recommendation did the trick:
> 
> * Create a virtualenv:
>   
>   * `virtualenv -p /usr/bin/python3 ~/.ve/sensorhat`
>   * `source ~/.ve/sensorhat/bin/activate`
> * Clone [RTIMULib](https://github.com/RPi-Distro/RTIMULib/):
>   
>   * `git clone https://github.com/RPi-Distro/RTIMULib/ RTIMU`
>   * `cd RTIMU/Linux/python`
> * Follow instruction in [RTIMULib/Linux/python](https://github.com/RPi-Distro/RTIMULib/blob/master/Linux/python/README.md):
>   
>   * `sudo apt install python3-dev`
>   * `python setup.py build`
>   * `python setup.py install`
> * Install `libopenjp2-7`:
>   
>   * `sudo apt install libopenjp2-7`
> * Install `sense-hat`:
>   
>   * `sudo apt install sense-hat`
> 
> That should do the trick. Some code like this should work:
> 
> ```python
> from sense_hat import SenseHat
> 
> sense = SenseHat()
> print('temperature', float(sense.get_temperature()))
> print('humidity', float(sense.get_humidity()))
> print('pressure', float(sense.get_pressure()))
> ```

This helped a lot, thank you so much 👍

Aonodensetsu, 2025-03-29:
this seems to be an issue now with a clean install of raspbian 64bit, installing the `git+https://github.com/RPi-Distro/RTIMULib@V7.2.1#subdirectory=Linux/python` version works

bsimmo, 2025-03-29:
> this seems to be an issue now with a clean install of raspbian 64bit, installing the `git+https://github.com/RPi-Distro/RTIMULib@V7.2.1#subdirectory=Linux/python` version works

Raspbian is 32bit (armhf), I assume you mean Raspberry Pi OS and probably the Debian 'Bookworm' version.

It just works for me on Raspberry Pi OS 64bit Desktop version. At least last time I tried, during initial testing of the 'Bookworm'
Are you starting with the lite version?

Aonodensetsu, 2025-03-30:
nope, `(recommended)` option, fresh install on pi with a sense hat already attached, i am connecting over ssh but desktop is installed

one kind of weird thing i am doing is running python through `marimo`, a jupyter-like, the setup is a venv with all the typical dependencies + marimo and the marimo ui open to a testfile with the sense hat import, which shows an rtimu error until i changed the version to git