astro-pi/python-sense-hat #59: OSError: Cannot detect Rpi-Sense FB device

brunisgirl, 2016-11-17:
I am using raspberry pi 3 and sense hat and get the above error.
I have tried everything in this link: 
https://github.com/RPi-Distro/python-sense-hat/issues/38

I still get the error. The LEDs go on when I connect the sense hat and off afterwards.
From demsg I get:
rpi-sense: 1-0046: Read from reg 240 failed
rpi-sense: probe of 1-0046 failed with error -22

Does this mean my sense hat is "fried"? (I forgot to add the legs before mounting the sense hat)

I appreciate any help.

Sahar

copilco, 2016-12-18:
Exactly the same situation. 
RTIMULib works well 
The sensehat works well in rpi 2.

brunisgirl, 2016-12-24:
I had to buy a new sensehat and now it works, I am afraid you might have fired it up as well.

mmodlin, 2016-12-30:
I have done everything that every other issue has resolved.  Including re-flashing the SD card with the latest version of Jessie and updating the firmware.  I am using the Raspberry pi 3. RTiMULib returns values.  I get the same results from dmesg.  When running a sample file I receive the same error message.  I hope someone can help!

XECDesign, 2017-02-28:
If you are seeing the issue and have tried to fix it, please report what you have tried, starting with a clean raspbian install. If it doesn't "just work" and it has never been plugged in the wrong way or otherwise short circuited , I would recommend returning it.

mushu999, 2017-06-04:
Did you try this? Add
dtoverlay=rpi-sense
in /boot/config.txt to force sense HAT detection.
Might be required if you have anything else plugged in...see:
[ https://www.raspberrypi.org/forums/viewtopic.php?f=104&t=133244 ]

closed 2017-09-25 by XECDesign