astro-pi/python-sense-hat #62: OSError: Cannot detect RPi-Sense FB device

kbraspi, 2016-12-21:
I am trying to work on a school project using sense hat. I am stuck at this point and have tried everything mentioned in the forums, not sure what else to do, any help is really appreciated.

Some information below:

uname -a
Linux raspberrypi 4.4.34+ #930 Wed Nov 23 15:12:30 GMT 2016 armv6l GNU/Linux

RTIMULibDrive11
Settings file RTIMULib.ini loaded
No IMU detected
Using fusion algorithm RTQF
No IMU found
i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         

Tried running a simple "Hello" program and get this cannot detect rpi-sense message

XECDesign, 2017-02-28:
If it doesn't "just work" and it has never been plugged in the wrong way or otherwise short circuited , I would recommend returning it.

mushu999, 2017-06-04:
Did you try this? Add 
dtoverlay=rpi-sense
in /boot/config.txt to force sense HAT detection.
Might be required if you have anything else plugged in...see:
[ https://www.raspberrypi.org/forums/viewtopic.php?f=104&t=133244 ]

closed 2017-09-25 by XECDesign