astro-pi/python-sense-hat #65: SenseHat PiFace2 conflict

framboise314, 2017-01-25:
Hello
I am a [french blogger](http://www.framboise314.fr). One of my readers reports a problem when using a Piface Digital 2 (as input ror command buttons) togather with the SenseHat. 
The issue appears when using Python interrupt to read buttons status. When pushing for the first time a button this fires an interrupt and the Led mattrix on the SenseHat displays the same pattern as when powering the Raspberry Pi. From there nothing, no program seems to be able to use the SenseHat. Only a power off/on is able to make SenseHat usable again.
I could not replicate this because i can not connect my PiFace2 and my SenseHat as did my reader with IO-SHIM.
I look for a possible cause of this problem. The only thing I found on [ModMyPi web page](https://www.modmypi.com/raspberry-pi/breakout-boards/raspberry-pi-(official)/raspberry-pi-sense-hat) is that 
_As a fun bonus mode, the SPI peripheral on the Atmel has been hooked up to the Pi’s SPI interface – you can reprogram your HAT in the field! We use this method to get the firmware into the Atmel during production test – and we leave it unprotected so you can substitute the stock firmware to get it to do whatever you want._
Is ther a way to disable this if it is the cause of the conflict ?
best regards
francois
www.framboise314.fr

XECDesign, 2017-01-26:
Stacking HATs with other HATs or other peripherals is not officially supported. The schematic and firmware source code are available for those who want to dive a little deeper, but it would be unreasonable for us to try to resolve all such conflicts, especially without schematics for the other board.

framboise314, 2017-01-26:
ok I understand
thank you for answering
best regards
françois

closed 2017-01-26 by framboise314