astro-pi/python-sense-hat #66: Another problematic pisense

ways, 2017-02-10:
What an incredible cute, but picky device.

I tried for a long time to get it working with a Pi 3, without luck.

On a Pi2 (rev1.2) it works sometimes. That is, I've had it working for a few seconds with one of the examples, but then it freezes, and dmesg shows read errors. It's frozen until I pull power.

Most times I boot up, the rainbow color stays, and dmesg shows:

> pi@pisense:~ $ dmesg |grep -i sense
> [    7.451905] rpi-sense 1-0046: Read from reg 240 failed
> [    7.451983] rpi-sense: probe of 1-0046 failed with error -22

I've tried to reflash eeprom, but can never get past "2. Disable write protection:". I just get an IO error. So I can read out the eeprom, but not write new.

XECDesign, 2017-02-28:
They definitely shouldn't be picky. A working sense hat 'just works' and if it has arrived in a state there it only work intermittently, I would recommend returning it.

ways, 2017-03-01:
Ok, thanks. Will return.

closed 2017-03-01 by ways