astro-pi/python-sense-hat #67: LED Brightness values of below 48 not lighting LEDs

MarkHutchy, 2017-02-13:
I have found that any R G B value of below 48 (or 8 in low light mode, which perhaps counter-intuitively, is to my eye about the same brightness as 48 in non-low light mode), is this an issue anyone else has noticed, I have tested this with two sense_hats on three different Raspberry Pi (an A, a 2B and a 3)
Please let me know if there is anything else you wish for me to try but I haven't seen this issue or a fix mentioned elsewhere
Thank you.
Mark

eah13, 2017-02-13:
This is how the module was designed.  Perhaps someone else can comment on exactly why this is the case, but for low_light and normal, there is a threshold below which the LEDs are not lit at all (and you've correctly determined what those thresholds are through your testing!)

MarkHutchy, 2017-02-13:
I understand there is sometimes a discrepancy between what value you have supplied to the LED and the value actually "displayed" (going on the note here - http://pythonhosted.org/sense-hat/api/#get_pixels ) but at no point have I seen anything to state that the RGB values that will be "displayed" to the LEDs is 48 (or 8) - 255, it is always stated to be 0-255. putting it another way, the range starting at 48 is excluding almost a fifth of the available range of brightness levels, and probably the most important for some projects.

XECDesign, 2017-02-13:
You can manually adjust the gamma curve. If you have a linear curve, the top half will all look the same when compared to the lower half. The curve is the way it is because you need to 'stretch' the lower end out.

MarkHutchy, 2017-02-14:
Ok, so I have now after looking into it a little further discovered that there are 32 possible values for brightness of LED, "0" - off to "31" full brightness, the default gamma setting has the first 6 values as zero which is why it doesn't light up till when it does.
sense.low_light = False
print (sense.gamma)
[0, 0, 0, 0, 0, 0, 1, 1, 2, 2, 3, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 14, 15, 17, 18, 20, 21, 23, 25, 27, 29, 31]
sense.low_light = True
print (sense.gamma)
[0, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 2, 2, 2, 3, 3, 3, 4, 4, 5, 5, 6, 6, 7, 7, 8, 8, 9, 10, 10]
Thanks for making me look into it a bit more, it's just a shame that there's not a dimmer setting for the lower end. Even at the lowest brightness (1 rather than zero) it still lights up the room at night. Will need to look into getting some sort of cover / diffuser.

MarkHutchy, 2017-02-14:
I'm sure someone could've explained that much better than I have or can, but reading here http://pythonhosted.org/sense-hat/api/#gamma has now helped me understand what's going on. Thanks XECDesign and eah13 for your comments.

eah13, 2017-02-14:
I admire the persistence!

> are 32 possible values for brightness of LED, "0" - off to "31" full brightness

That's the explanation I knew was out there somewhere! I suppose the module coders had to choose between blank LEDs at the lower end of the range, capping total brightness, and inserting random places where the step between values wouldn't increase the brightness. Still, seems like getting above 1 sooner would've been nice. 

Diving in to hardware and software always yields little gems like this. I love how wonderfully contingent everything always turns out to be. Thanks for sharing!

XECDesign, 2017-02-23:
Alright to close?

closed 2017-02-24 by MarkHutchy

MarkHutchy, 2017-02-24:
I have decided that using a seemingly smaller range will allow for a greater variance in brightness over the selected range, as changes in brightnesses above a certain level (gamma value of around 15) are almost indistinguishable to the eye meaning that around half of the range (when set to a linear gamma "curve") is pretty much full brightness. Again I hope that makes sense.