astro-pi/python-sense-hat #70: Raspberry Pi Sense-Hat LED Matrix stays lit to rainbow pattern, and inaccessible.

Unimprezzed, 2017-03-30:
I got the Raspberry Pi Sense-Hat to work on an earlier build of Raspian, and was even able to program some simple programs using it. Now, the LED Matrix remains on, permanently showing the rainbow test pattern. 

I try to clear it in Python 3, and I get the following when I try to instantiate the object: 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/sense_hat/sense_hat.py", line 39, in __init__
  raise OSError('Cannot detect %s device' % self.SENSE_HAT_FB_NAME) 
OSError: Cannot detect RPi-Sense FB device

I've checked the connections to the RPi and Sense-Hat, and they seem secure. This is in fact been an ongoing problem, as I ended up exchanging both a perfectly good RPi3 and Sense-Hat for new ones, only to have the same issue. 

Any input would be appreciated.

XECDesign, 2017-03-30:
What has happened between the time it was working and not?
What's the output of i2cdetect -y 1?

Unimprezzed, 2017-03-30:
Thank you for the reply. Between the time where the system worked properly and when it started to give me trouble was a system update. Basically, 
`sudo apt-get update`
`sudo apt-get upgrade`
`sudo apt-get dist-upgrade`

I'm afraid that entering the output of that command gets messed up in the text due to formatting isssues, so I've attached the textfile containing output. 
[out.txt](https://github.com/RPi-Distro/python-sense-hat/files/883530/out.txt)

XECDesign, 2017-03-30:
Could you also post the output of `raspi-gpio get`?

Unimprezzed, 2017-03-30:
Sure thing. 
Again in the text file to skip formatting. 
[gpiooutput.txt](https://github.com/RPi-Distro/python-sense-hat/files/883603/gpiooutput.txt)

XECDesign, 2017-03-30:
Is there anything else connected to GPIO? It looks like none of the devices are acknowledging that they're there. This could happen if something else was connected and was pulling the pin low.

It might be worth trying the earlier build again. If that doesn't work anymore, then you'll know it's a hardware problem.

Unimprezzed, 2017-03-30:
Just a few buttons. I followed the instructions provided at [https://www.raspberrypi.org/learning/astro-pi-flight-case/worksheet2/](url), and after testing, the buttons and camera are all functional. 

How do I revert to an older build of Raspian?

XECDesign, 2017-03-30:
An archive of all the images can be found here:
http://downloads.raspberrypi.org/raspbian/images/

Another possibility is that the HAT itself was damaged or the pin on the pi was damaged. That could've happened if there was a short, for example. If you have any other I2C devices you could test, that would be useful information, but I suspect you might not have anything suitable lying around.

Unimprezzed, 2017-03-30:
Unfortunately, I don't have any other I2C devices. I've tried this on two SenseHats, on two separate RPi's, and I doubt the possibility that it's a hardware issue. 

Nevertheless, I tried loading the library, and the issue is still not resolved. 

I have a multi-meter. Is there any guide for testing the pins with a regular multimeter?

Unimprezzed, 2017-03-31:
Just saw your edit. Here's the output of the two commands you wanted to see. This is from the 
[uname.txt](https://github.com/RPi-Distro/python-sense-hat/files/884104/uname.txt)
[vcgencmd.txt](https://github.com/RPi-Distro/python-sense-hat/files/884105/vcgencmd.txt)

XECDesign, 2017-03-31:
If we've ruled out software (by returning to your previous setup), and you're saying it couldn't possibly be hardware, what does that leave?

If you start with a fresh, untouched card with only power and the HAT connected, you should see the rainbow pattern go on and then turn off when the kernel loads the relevant module. If that doesn't happen, that certainly points to a hardware issue.

Using the multimeter, you could arm yourself with the datasheet of one of the i2c devices and the sense hat schematic (https://www.raspberrypi.org/wp-content/uploads/2015/08/Sense-HAT-V1_0.pdf), you could do some basic troubleshooting. While it's running, carefully check the main voltage rails. When it's disconnected, check the connections and that there are no shorts to ground on the sda/scl lines. To check what's really happening, you'd want to use a scope and/or a logic analyzer.

Unimprezzed, 2017-04-08:
You were right, it was a hardware issue. Except not with the boards. 

It was with the extended 23-way pin header I was using. Either it was a dud or it was poorly made, because I swapped it out with a more sturdy extended header from Adafruit, and now it's working just fine. Thanks for your input, and I'm closing the issue.

closed 2017-04-10 by XECDesign