astro-pi/python-sense-hat #71: OSError: Cannot detect RPi-Sense FB device

FunkMast3rFlex, 2017-04-02:
I know this seems to be an common issue but I have looked over all the previous threads and nothing has worked for me. Similar I believe that my SenseHat may just be fried but I am still fairly new with the Raspberry Pi so I leave that diagnosis to professionals. Like most I too get the error prompt:
```
File "/usr/lib/python3/dist-packages/sense_hat/sense_hat.py", line 39, in __init__
    raise OSError('Cannot detect %s device' % self.SENSE_HAT_FB_NAME)
OSError: Cannot detect RPi-Sense FB device
```
I have done some troubleshooting myself by switching out Sense Hats with a friend and their Sense Hat works fine so I know that the problem is my SenseHat. The Sense Hat LED board lights up like normal when connected to power but will continue to stay on even after boot. Aside from this I have also takes the following measurses:
- Enabling 12C
- Rebooting and Re-Seeding the SenseHat
- sudo apt-get update
- sudo apt-get upgrade
- sudo apt-get dist-upgrade
- sudo rpi-update
- sudo RTIMULibDrive11
```
Settings file not found. Using defaults and creating settings file
Detected LSM9DS1 at standard/standard address
Using fusion algorithm RTQF
Detected LPS25H at standard address
Detected HTS221 at standard address
min/max compass calibration not in use
Ellipsoid compass calibration not in use
Accel calibration not in use
LSM9DS1 init complete
Sample rate 0: : roll:-11.688202, pitch:-4.321792, yaw:-74.485576
Pressure: 1023.1, height above sea level: -81.7, temperature: 37.2, humidity: 39.9
```

I had read that enabling **device tree** in `raspi-config` often did the trick but I did not see that option. I had planned to `nano /boot/config.txt` and add the line `device_tree=` but once again I would rather leave that to your judgment. Same goes for setting `dtoverlay=rpi-sense` and `dtdebug=1`.

I have already gone ahead and ran the following codes:
```
sudo apt-get install device-tree-compiler
uname -a
vcgencmd version
sudo dtc -I fs /proc/device-tree
```

uname -a
```
Linux RasPi 4.9.19-v7+ #983 SMP Thu Mar 30 14:46:28 BST 2017 armv7l GNU/Linux
```

vcgencmd version
```
Mar 30 2017 14:52:31 
Copyright (c) 2012 Broadcom
version 110f1df714664905d565d50b215c22c988b90a0a (clean) (release)
```

sudo dtc -I fs /proc/device-tree
```
Warning (unit_address_vs_reg): Node /soc has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /soc/vchiq has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /soc/gpiomem has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /soc/local_intc has a reg or ranges property, but no unit name
Warning (unit_address_vs_reg): Node /memory has a reg or ranges property, but no unit name
/dts-v1/;

/ {
	compatible = "brcm,bcm2837", "brcm,bcm2836";
	serial-number = "000000001be30aac";
	model = "Raspberry Pi 3 Model B Rev 1.2";
	memreserve = <0x37000000 0x8000000>;
	interrupt-parent = <0x1>;
	#address-cells = <0x1>;
	#size-cells = <0x1>;

	clocks {
		compatible = "simple-bus";
[... 848 lines cut ...]
			reg = <0x2>;
			clock-frequency = <0x47868c00>;
		};

		cpu@0 {
			compatible = "arm,cortex-a7";
			device_type = "cpu";
			phandle = <0x1d>;
			reg = <0x0>;
			clock-frequency = <0x47868c00>;
		};
	};

	fixedregulator_5v0 {
		compatible = "regulator-fixed";
		phandle = <0x47>;
		regulator-min-microvolt = <0x4c4b40>;
		regulator-max-microvolt = <0x4c4b40>;
		regulator-always-on;
		regulator-name = "5v0";
	};
};
```

I apologize if it may seem asinine dealing with the same issues, so if there is anything I can do to speed the process please let me know. Thank You for your support.

XECDesign, 2017-04-03:
The best thing to try is to attach the HAT and boot a Raspbian clean image with no changes. If that doesn't just work, you could try adding `dtoverlay=rpi-sense` to config.txt. If that still doesn't work and you've never plugged it in incorrectly, I would say that the board should be returned.

If you're feeling adventurous, you could try re-programming the Atmel. If you're going to get a replacement, it might be safer to not do that.

FunkMast3rFlex, 2017-04-04:
Adding `dtoverlay=rpi-sense` did the job. Thank You, sorry to be the one to bring up another one of the same issues to again have a simple fix. Before seeing this I did however talk a friend into giving me his Sense Hat. I plan on reprogramming the Atmel anyway even though it works and if anything goes wrong I will have a backup. Is there anything that I should know before trying it out?

XECDesign, 2017-04-06:
The issue here seems to be with the EEPROM, not the Atmel. When the pi boots, it should read the EEPROM, which would tell it that the HAT is present. If you would like to do it anyway, the instructions can be found here: https://github.com/raspberrypi/documentation/blob/master/hardware/sense-hat/README.md

They might be a little out of date, but I should be able to help if you run into trouble.

closed 2017-04-26 by XECDesign