astro-pi/python-sense-hat #73: Strange behavior set_pixel

paulocanedo, 2017-05-11:
Hi,

```python
sense.clear()
sense.set_pixel(0, 0, 255, 0, 0)
```

When I use above code, all the LEDs in column 0 turns red, instead of just one LED, how can I find where the bug is? python lib, SenseHat firmware, board connection?

XECDesign, 2017-05-11:
Could you run `sudo dd if=/dev/fb1 of=test.rgb565` and then upload test.rgb565?

paulocanedo, 2017-05-12:
here is @XECDesign 
[test.zip](https://github.com/RPi-Distro/python-sense-hat/files/996137/test.zip)

XECDesign, 2017-05-12:
So it's not the python lib. Given that I can't reproduce the problem, it's hard to investigate the rest. Are you overclocking or changing any of the timings at all?

paulocanedo, 2017-05-12:
No over clock or any kind of hack... i have a raspbian fresh install and modified /boot/confia adding dtoverlay = sensehat . Thanks for your help, at least I know the problem isn't python lib.

XECDesign, 2017-05-12:
Do you have another pi or another sense hat you could try? Do you have a logic analyser or a scope?

closed 2017-09-25 by XECDesign