astro-pi/python-sense-hat #74: OSError: Cannot detect RPi-Sense FB device

somucal, 2017-05-17:
**Code:**
`from sense_hat import SenseHat
sense = SenseHat()`

**ERROR:**
pi@raspberrypi:~ $ python3 test.py 
Traceback (most recent call last):
  File "test.py", line 3, in <module>
    sense = SenseHat()
  File "/usr/lib/python3/dist-packages/sense_hat/sense_hat.py", line 39, in __init__
    raise OSError('Cannot detect %s device' % self.SENSE_HAT_FB_NAME)
OSError: Cannot detect RPi-Sense FB device

**Symtoms:**
When i switch on the pi 3 Model B with sense hat plugged in, all the LED's are illuminated and stays that way as long as the pi is on. 

What I've tried 
1. Enabled I2C from Raspberry Pi Configuration
2. Ran the following 
sudo apt-get update
sudo apt-get upgrade
sudo apt-get install sense-hat
sudo pip3 install pillow

Let me know -

1) Is the Sense Hat defective?
2) Is the Pi defective?
3) How do we register the SenseHat in Pi configuration/devices? Any way to check that?

Appreciate any help. Thanks.

XECDesign, 2017-05-18:
Could you please post the output of `i2cdetect -y 1`, `dmesg` and `raspi-gpio get`?

somucal, 2017-05-18:
pi@raspberrypi:~ $ clear
pi@raspberrypi:~ $ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
**-----------------------------------------------------------------------------------------------**
pi@raspberrypi:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.50-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #970 SMP Mon Feb 20 19:18:29 GMT 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 808c5040, node_mem_map b9fa6000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] [bcm2709_smp_init_cpus] enter (9520->f3003010)
[    0.000000] [bcm2709_smp_init_cpus] ncores=4
[    0.000000] PERCPU: Embedded 13 pages/cpu @b9f62000 s22592 r8192 d22464 u53248
[    0.000000] pcpu-alloc: s22592 r8192 d22464 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 dma.dmachans=0x7f35 bcm2709.boardrev=0xa22082 bcm2709.serial=0xc578c78e bcm2709.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:78:C7:8E vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=a3523b56-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 939064K/966656K available (6357K kernel code, 432K rwdata, 1716K rodata, 476K init, 764K bss, 19400K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x807ea73c   (8074 kB)
      .init : 0x807eb000 - 0x80862000   ( 476 kB)
      .data : 0x80862000 - 0x808ce350   ( 433 kB)
       .bss : 0x808d1000 - 0x809901ec   ( 765 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000026] Switching to timer-based delay loop, resolution 52ns
[    0.000288] Console: colour dummy device 80x30
[    0.000487] console [tty1] enabled
[    0.000510] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000530] pid_max: default: 32768 minimum: 301
[    0.000844] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000858] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001783] Disabling cpuset control group subsystem
[    0.001814] Initializing cgroup subsys io
[    0.001844] Initializing cgroup subsys memory
[    0.001887] Initializing cgroup subsys devices
[    0.001907] Initializing cgroup subsys freezer
[    0.001928] Initializing cgroup subsys net_cls
[    0.001977] CPU: Testing write buffer coherency: ok
[... 348 lines cut ...]
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=1 fsel=7 alt=3 func=TXD0
  GPIO 33: level=1 fsel=7 alt=3 func=RXD0
  GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
  GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
  GPIO 40: level=1 fsel=4 alt=0 func=PWM0
  GPIO 41: level=0 fsel=4 alt=0 func=PWM1
  GPIO 42: level=1 fsel=4 alt=0 func=GPCLK1
  GPIO 43: level=0 fsel=4 alt=0 func=GPCLK2
  GPIO 44: level=1 fsel=5 alt=1 func=SDA0
  GPIO 45: level=1 fsel=5 alt=1 func=SCL0
BANK2 (GPIO 46 to 53):
  GPIO 46: level=1 fsel=0 alt=  func=INPUT
  GPIO 47: level=1 fsel=0 alt=  func=INPUT
  GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
  GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
  GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
  GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
  GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
  GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3
pi@raspberrypi:~ $

XECDesign, 2017-05-18:
How is the HAT connected to the pi? Are you using the expansion header that comes with the sense hat to raise it to the right level? Is there anything else between the pi and the HAT? Is anything else connected to the GPIO header?

somucal, 2017-05-18:
Its connected directly to the pi.

somucal, 2017-05-18:
[image]

XECDesign, 2017-05-18:
I can't tell from the photo whether the expansion header is there. It looks tilted like it's not.
http://spellfoundry.com/wp/wp-content/uploads/2016/05/GPIO-Header-Short-Pins-1.jpg

somucal, 2017-05-18:
Its not there, I removed it.

XECDesign, 2017-05-18:
It's required for a solid connection

somucal, 2017-05-18:
okay, I can check that later today and report back. Thanks for your quick help.

somucal, 2017-05-19:
So I did remove the sense hat and plugged in with the expansion header. It did work and the lights are no longer all on. They came on when I connected power, half red and half otherwise and then went away. However the error on running the python script is same.

here is the output of the commands you asked. Let me know what I am doing wrong.

Also, I am using RPi3 Model B with Raspbian Jessie with Pixel as OS.

Thanks and Appreciate your help.

**pi@raspberrypi:~ $ i2cdetect -y 1**
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

**pi@raspberrypi:~ $ dmesg**
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.28-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #999 SMP Tue May 16 14:03:19 BST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x3a800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 241664
[    0.000000] free_area_init_node: node 0, pgdat 80c6edc0, node_mem_map b9faa000
[    0.000000]   Normal zone: 2124 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 241664 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @b9f63000 s25600 r8192 d23552 u57344
[    0.000000] pcpu-alloc: s25600 r8192 d23552 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 239540
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3dc00000 vc_mem.mem_size=0x3f000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=a3523b56-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 936296K/966656K available (7168K kernel code, 483K rwdata, 2012K rodata, 1024K init, 778K bss, 22168K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
    vector  : 0xffff0000 - 0xffff1000   (   4 kB)
    fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
    vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
    lowmem  : 0x80000000 - 0xbb000000   ( 944 MB)
    modules : 0x7f000000 - 0x80000000   (  16 MB)
      .text : 0x80008000 - 0x80800000   (8160 kB)
      .init : 0x80b00000 - 0x80c00000   (1024 kB)
      .data : 0x80c00000 - 0x80c78f94   ( 484 kB)
       .bss : 0x80c7a000 - 0x80d3ca64   ( 779 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000008] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000020] Switching to timer-based delay loop, resolution 52ns
[    0.000318] Console: colour dummy device 80x30
[    0.000337] console [tty1] enabled
[    0.000359] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000376] pid_max: default: 32768 minimum: 301
[    0.000711] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000720] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001710] Disabling cpuset control group subsystem
[    0.001870] CPU: Testing write buffer coherency: ok
[    0.001910] ftrace: allocating 22386 entries in 66 pages
[    0.054851] CPU0: update cpu_capacity 1024
[... 249 lines cut ...]
  GPIO 31: level=0 fsel=0 alt=  func=INPUT
  GPIO 32: level=1 fsel=7 alt=3 func=TXD0
  GPIO 33: level=1 fsel=7 alt=3 func=RXD0
  GPIO 34: level=1 fsel=7 alt=3 func=SD1_CLK
  GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
  GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
  GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
  GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
  GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
  GPIO 40: level=1 fsel=4 alt=0 func=PWM0
  GPIO 41: level=0 fsel=4 alt=0 func=PWM1
  GPIO 42: level=0 fsel=4 alt=0 func=GPCLK1
  GPIO 43: level=1 fsel=4 alt=0 func=GPCLK2
  GPIO 44: level=1 fsel=5 alt=1 func=SDA0
  GPIO 45: level=1 fsel=5 alt=1 func=SCL0
BANK2 (GPIO 46 to 53):
  GPIO 46: level=1 fsel=0 alt=  func=INPUT
  GPIO 47: level=1 fsel=0 alt=  func=INPUT
  GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
  GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
  GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
  GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
  GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
  GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3
pi@raspberrypi:~ $

XECDesign, 2017-05-19:
It looks like you're getting a partial connection now. I worry that the connector on the sense hat may have been damaged when it was connected at an angle and now it's not making proper contact.

somucal, 2017-05-19:
So should I replace the Sense Hat? If that helps I can try that

somucal, 2017-05-19:
I removed it and tried to fit it back in, it worked! Let me run a few more tests. Thx for your support.

XECDesign, 2017-05-19:
Glad it works now.

closed 2017-05-19 by XECDesign