astro-pi/python-sense-hat #76: Cannot detect RPi-Sense FB device

teuteuguy, 2017-07-05:
I'm trying to use the Sense Hat in the context of a containerized environment for execution.
(AWS Greengrass).

I get the following error: Cannot detect RPi-Sense FB device

Is there a way to package sense-hat locally ?
via: pip install sense-hat -t .

Problem is when trying to do that, it fails on an error: 
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-JdcXGG/numpy/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-V_cuj6-record/install-record.txt --single-version-externally-managed --compile --home=/tmp/tmpYI3tTo failed with error code 1 in /tmp/pip-build-JdcXGG/numpy
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 248, in main
    return command.main(cmd_args)
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 161, in main
    text = '\n'.join(complete_log)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 72: ordinal not in range(128)

XECDesign, 2017-07-05:
You could if you update pip, but that won't fix the problem. How is the sense hat connected to the pi? Could you try reseating it a few times to make sure you have good contacts on all pins?

teuteuguy, 2017-07-05:
No need to re-seat it. I've tried.

Also, when trying to run the code outside of the context of this containerized environment, it works fine. So something is odd.

teuteuguy, 2017-07-05:
Arggggg wasn't supposed to close it

XECDesign, 2017-07-05:
Does that environment have access to /dev/fb1 ?

teuteuguy, 2017-07-05:
That's something I don't know. And i don't know how to find out.

How can I from the python script expose a lot of information about the environment ?

XECDesign, 2017-07-05:
Okay, that's getting into specifics of python and containers. There are 100s of ways of doing this and I am not an expert. This might be better as a forum question.

XECDesign, 2017-07-11:
Closing it as it's either a generic python question or is specific to AWS Greengrass.

closed 2017-07-11 by XECDesign