astro-pi/python-sense-hat #77: OSError: Cannot detect RPi-Sense FB device

canxerLX, 2017-08-02:
Hi there ! Recently receive my rpi together with rpi. Using raspbian and installed sense-hat according to documentation. Still receiving this issue while trying the examples provided for sense-hat.

was looking through **"OSError: Cannot detect RPi-Sense FB device #28"** and I tried the following command:

**_sudo apt-get install device-tree-compiler_**

and I received this output:

**_pi@raspberrypi:~ $ sudo apt-get install device-tree-complier
Reading package lists... Done
Building dependency tree       
Reading state information... Done
**Unable to locate package device-tree-complier**_**

could it be **"unable to locate package device-tree-complier"** the problem? if so, how can I solve it?

Thanks ! on the side note I am working on my final year project using the sense-hat :)

XECDesign, 2017-08-02:
Sorry to hear you're having trouble with it. How is the HAT connected to the pi? This usually seems to be caused by poor contact or an incorrectly fitted HAT. Avoid removing or attaching it while the pi is powered on. 

Also note that it's compiler, not complier (typo).

canxerLX, 2017-08-03:
[image]

This is how I connect my HAT to the pi. Am I supposed to push it all the way down? cause the last i tried it didnt work and I was afraid I might push it way down. So the inability to locate the device tree compiler is due to the loose connection of the HAT?

XECDesign, 2017-08-03:
Yes, you do need to push it all the way down, and I believe it should be okay after that. If it feels like you're having to force it, check that all the pins are lined up and none are getting bent.

The reason that you were unable to install device-tree-compiler is that you're written it wrong
You typed 'device-tree-complier' instead of 'device-tree-compiler'. I don't think you'll need it anyway.

canxerLX, 2017-08-03:
I've pushed it down all the way. Still didnt work. here's some of the output from my terminal:

**sudo apt-get install device-tree-compiler**
Reading package lists... Done
Building dependency tree       
Reading state information... Done
device-tree-compiler is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

**uname -a**
Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

**vcgencmd version**
Jul  3 2017 14:20:31 
Copyright (c) 2012 Broadcom
version 4139c62f14cafdb7d918a3eaa0dbd68cf434e0d8 (tainted) (release)

**ls /dev/fb***
/dev/fb0

[dmesg.txt](https://github.com/RPi-Distro/python-sense-hat/files/1197285/dmesg.txt)
[sudo dtc -I fs (slash)proc(slash)device-tree.txt](https://github.com/RPi-Distro/python-sense-hat/files/1197286/sudo.dtc.-I.fs.slash.proc.slash.device-tree.txt)

the device-tree-compiler is fine now. I am guessing there is something wrong with **dmesg** and **sudo dtc -I fs /proc/device-tree** because there are some warnings and unsuccessful messages.

P.s. I am learning everything from scratch. please be patient with me :P

XECDesign, 2017-08-07:
Sorry about the late reply.

What that output tells us is that the HAT is not seen. It could be because the HAT is faulty or because there is still an issue with the contacts.

I would shut down the pi, unplug the power, take the HAT off and put it back on a few times to make sure it's properly seated. If that doesn't work, I'd try adding `dtoverlay=rpi-sense` to `/boot/config.txt` and rebooting. That would bypass the auto-detection and act as if the HAT is there even if it doesn't look like it. If that still doesn't work, I would ask for a replacement from the seller.

Zam95, 2017-08-28:
@XECDesign Hello, I am a student and I have the same problem to use my sense hat for a project, I already read many answers that you wrote but I cant to make it work, could you help me please? I have a RPI 3 and the error says 

OSError: Cannot detect Rpi-Sense FB device

XECDesign, 2017-08-31:
Hi @Zam95, if the information in this issue and other issues does not help, you may have to get the HAT replaced. If it's attached correctly and has good contact with all of the pins, but isn't working while running Raspbian, then there's something wrong with the hardware.

Zam95, 2017-09-13:
Hi @XECDesign I try to make sense hat work with 2 diferent raspberrys. In the raspberry pi3 the leds only flash in white when I connect raspberry to the power, in the raspberry pi2 the leds always are on in rainbow until I disconect the raspberry.

I put `sudo i2cdetect -y 1` and the output is:

```
pi@raspberrypi:~ $ sudo i2cdetect -y 1
     
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 

10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- -- 

20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f 

60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- -- 

70: -- -- -- -- -- -- -- --

```
A forum moderator tell me AVR is not detected and give me the next link: https://github.com/raspberrypi/rpi-sense

Do you have instrucctions for the process? I see files but i dont understand, sorry I am new in the Raspberry world :(
Thanks for your attention and sorry for my bad english

XECDesign, 2017-09-14:
I don't have instructions, but off the top of my head:
On a pi with the sense hat, enable SPI in raspi-config, clone that repo, install dependencies, run `make` and `make flash`. 

This is getting more into AVR territory than Raspberry Pi.

Zam95, 2017-09-14:
Hi @XECDesign  Thanks for your reply, I already clone that repo, SPI is enabled, but what is "install dependencies"?

When I put `make` inn terminal the output is  `make: *** No targets specified and no makefile found.    Stop.`

Sorry I am really really n00b but i need make it work the Sense hat

XECDesign, 2017-09-14:
> Hi @XECDesign Thanks for your reply, I already clone that repo, SPI is enabled, but what is "install dependencies"?

`sudo apt-get install gcc-avr avr-libc avrdude`

> When I put make inn terminal the output is make: *** No targets specified and no makefile found. Stop.
You need to run it from the rpi-sense directory that you cloned earlier.

Zam95, 2017-09-14:
Thanks @XECDesign  I install dependencies, I run Make and the output is:

```
pi@raspberrypi:~/rpi-sense $ make
avr-gcc -g -Wall -Os -mmcu=attiny88 -c -o main.o main.c
avr-gcc -Wall -mmcu=attiny88 -c -o rpi-sense.o rpi-sense.S
avr-gcc -g -Wall -Os -mmcu=attiny88 -Wl,- Map,rpi-sense.map -o rpi-sense.elf main.o rpi-
sense.o
avr-objdump -h -S rpi-sense.elf &gt; rpi-sense.lst
avr-objcopy -j .text -j .data -O ihex rpi-sense.elf rpi-sense.hex
avr-objcopy -j .text -j .data -O binary rpi-sense.elf rpi-sense.bin
avr-objcopy -j .text -j .data -O srec rpi-sense.elf rpi-sense.srec
avr-objcopy -j .eeprom -- change-section- lma .eeprom=0 -O ihex rpi-sense.elf rpi-
sense_eeprom.hex \
|| { echo empty rpi-sense_eeprom.hex not generated; exit 0; }
avr-objcopy: -- change-section- lma .eeprom=0x00000000 never used
avr-objcopy -j .eeprom -- change-section- lma .eeprom=0 -O binary rpi-sense.elf rpi-
sense_eeprom.bin \
|| { echo empty rpi-sense_eeprom.bin not generated; exit 0; }
avr-objcopy: -- change-section- lma .eeprom=0x00000000 never used
avr-objcopy -j .eeprom -- change-section- lma .eeprom=0 -O srec rpi-sense.elf rpi-
sense_eeprom.srec \
|| { echo empty rpi-sense_eeprom.srec not generated; exit 0; }
avr-objcopy: -- change-section- lma .eeprom=0x00000000 never used
```
and `make flash`

```
pi@raspberrypi:~/rpi-sense $ make flash
sudo avrdude -c linuxspi -p t88 -P /dev/spidev0.0 -U flash:w:rpi-sense.hex
avrdude: AVR device initialized and ready to accept instructions
Reading | ################################################## | 100% 0.00s
avrdude: Device signature = 0x1e9311 (probably t88)
avrdude: NOTE: &quot;flash&quot; memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file &quot;rpi-sense.hex&quot;
avrdude: input file rpi-sense.hex auto detected as Intel Hex
avrdude: writing flash (1168 bytes):
Writing | ################################################## | 100% 0.24s
avrdude: 1168 bytes of flash written

avrdude: verifying flash memory against rpi-sense.hex:
avrdude: load data flash data from input file rpi-sense.hex:
avrdude: input file rpi-sense.hex auto detected as Intel Hex
avrdude: input file rpi-sense.hex contains 1168 bytes
avrdude: reading on-chip flash data:
Reading | ################################################## | 100% 0.16s
avrdude: verifying ...
avrdude: 1168 bytes of flash verified
avrdude: safemode: Fuses OK (E:FF, H:DF, L:EE)
avrdude done. Thank you.
pi@raspberrypi:~/rpi-sense $
```
But when I put `sudo i2cdetect -y 1` the same 

```
pi@raspberrypi:~ $ sudo i2cdetect -y 1
     
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 

10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- -- 

20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f 

60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- -- 

70: -- -- -- -- -- -- -- --
```

and OSError persist

XECDesign, 2017-09-14:
I am a little surprised it was even programmable. There's not much else you can do without access to a scope or logic analyser and an AVR JTAG debugger and that's certainly not something somebody could talk you through remotely.

bsimmo, 2017-09-14:
AVR was left open by design
'...As a fun bonus mode, the SPI peripheral on the Atmel has been hooked up
to the Pi’s SPI interface – you can reprogram your HAT in the field! We use
this method to get the firmware into the Atmel during production test – and
we leave it unprotected so you can substitute the stock firmware to get it
to do whatever you want. Seriously. First person to turn this sensor HAT
into a quadcopter controller HAT wins a cookie from me.'

Does the bottom of this page help ?
https://www.raspberrypi.org/documentation/hardware/sense-hat/

XECDesign, 2017-09-14:
@bsimmo, op has already re-programmed the AVR firmware and it has not helped the issue. The instructions on the page you linked are for flashing the EEPROM, which would not help.

bsimmo, 2017-09-14:
I though that but it's straight after the AVR heading.. that's why I asked
if it would help.

(I was manly saying why the AVR was open to write, though given it can be
written to why can it not see it (is that a different interface and what
you meant)).

I'd send it back personally/ask for a replacement (warranty), something's
not right obviously.

XECDesign, 2017-09-14:
Ah right. Programming is done over SPI. Although that part of it is clearly working, I2C communication is not.

Zam95, 2017-09-14:
Thanks @XECDesign  and @bsimmo I am very confused, the behavior of the Sense hat is weird. With the RPI3 just conect RPI to power the leds make a fast white blink and immediately turn off the leds

`sudo i2cdetect -y 1` the output is:

```
pi@raspberrypi:~ $ sudo i2cdetect -y 1
     
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 

10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- -- 

20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f 

60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- -- 

70: -- -- -- -- -- -- -- --
```

but in RPI2 just connect to the power the leds turn on in rainbow colors and never turn off until I disconnect

In RPI2 `sudo i2cdetect -y 1` the output is:

```
pi@raspberrypi:~ $ sudo i2cdetect -y 1
     
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 

10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

70: -- -- -- -- -- -- -- --

```
but `sudo i2cdetect -y 0` the output is: 

```
pi@raspberrypi:~ $ sudo i2cdetect -y 0
     
0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f

00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 

10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

50: 50 -- -- -- -- -- -- 58 -- -- -- -- -- -- -- -- 

60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 

70: -- -- -- -- -- -- -- --
```

The number "50" sometimes changes to "UU"

XECDesign, 2017-09-15:
When you run `i2cdetect -y 0`, you're looking devices on a different bus.

The different behaviour between devices is interesting. It means when the device is plugged into a pi 3, the firmware on the AVR is not able to turn on the LEDs and display the rainbow pattern. On the pi 2, it is. This is the first thing the AVR tries to do when it gets power.

How are you powering your pi? Do you ever see a lightning bolt or any other icon in the top right corner?

Zam95, 2017-09-19:
Hi @XECDesign my power supply give me 5V, 2000 mA, I remember at some point
a lightning appeared in RPI3

Do you know if the sense hat can be repair? Where? Or I have to throw it in
the trash and buy a new :(

Greetings from Mexico.

XECDesign, 2017-09-25:
It's not something you'd typically get repaired, but it should be covered by warranty.

closed 2017-09-25 by XECDesign