astro-pi/python-sense-hat #79: All sensors output 0.0

jwhendy, 2017-09-24:
Greetings,

I'm just getting started with the sense hat and am likely doing something wrong. I was working through [this guide](https://www.raspberrypi.org/learning/getting-started-with-the-sense-hat/worksheet/) and controlling the LEDs was going great. I moved on to try the accelerometer example and just get zeros.

```
$ cat acc-example.py
from sense_hat import SenseHat

sense = SenseHat()

while True:
    acceleration = sense.get_accelerometer_raw()
    x = acceleration['x']
    y = acceleration['y']
    z = acceleration['z']

    x=round(x, 0)
    y=round(y, 0)
    z=round(z, 0)

    print("x={0}, y={1}, z={2}".format(x, y, z))
```

When running the above, I just get this:
```
$ python acc-example.py 
x=0.0, y=0.0, z=0.0
x=0.0, y=0.0, z=0.0
x=0.0, y=0.0, z=0.0
x=0.0, y=0.0, z=0.0
...
```

I'm using [arch ARM](https://archlinuxarm.org/platforms/armv8/broadcom/raspberry-pi-3).

```
$ cat /boot/config.txt 
# See /boot/overlays/README for all available options

gpu_mem=128
initramfs initramfs-linux.img followkernel

start_file=start_x.elf
fixup_file=fixup_x.dat

dtparam=i2s=on
dtparam=spi=on
dtparam=i2c_arm=on
dtoverlay=rpi-sense

$ cat /etc/modules-load.d/raspberrypi.conf 
snd-bcm2835
i2c-dev
bcm2835-v4l2
```

Does anything look awry above, or would you have tips on finding out what's wrong?

jwhendy, 2017-09-24:
Solved. It was a permissions issue.

Default:
```
$ ls -l /dev/i2c-1 
crw------- 1 root root 89, 1 Aug 16 12:20 /dev/i2c-1
```

After implementing a `udev` rule:
```
$ cat /etc/udev/rules.d/99-i2c.rules 
KERNEL=="i2c-[0-7]",MODE="0666"

$ ls -l /dev/i2c-1 
crw-rw-rw- 1 root root 89, 1 Aug 16 12:20 /dev/i2c-1
```

jwhendy, 2017-09-24:
Aside/comment: in trying to troubleshoot this, there are a lot of examples online where `sudo` is suggested in an error message. Mine didn't feature this:

```
from sense_hat import SenseHat
import time
 
sense = SenseHat()
 
while True:
 
   t = sense.get_temperature()
   p = sense.get_pressure()
   h = sense.get_humidity()

   msg = "Temperature = %s, Pressure=%s, Humidity=%s" % (t,p,h)
   print(msg)
   time.sleep(0.5)
```

```
  File "scripts/sensors.py", line 8, in <module>
    t = sense.get_temperature()
  File "/usr/lib/python3.6/site-packages/sense_hat/sense_hat.py", line 606, in get_temperature
    return self.get_temperature_from_humidity()
  File "/usr/lib/python3.6/site-packages/sense_hat/sense_hat.py", line 582, in get_temperature_from_humidity
    self._init_humidity()  # Ensure humidity sensor is initialised
  File "/usr/lib/python3.6/site-packages/sense_hat/sense_hat.py", line 549, in _init_humidity
    raise OSError('Humidity Init Failed')
OSError: Humidity Init Failed
```

This led me to think it was an issue with my devices or modules vs. permissions. Finally it dawned on me to just try `sudo` anyway, and to my amazement it worked!

**Could the error message be improved to help users understand if it's device existence vs. permissions?**

bennuttall, 2018-03-23:
sudo shouldn't be needed if the Pi is configured correctly. I see you're using Python 3.6 which is not the standard in Raspbian. Are you not using Raspbian? How are you using Python 3.6?

The apt package `sense-hat` will make everything work. As long as it's installed, you can even use a virtualenv and just install the python libraries required - see my comments here: https://github.com/RPi-Distro/python-sense-hat/issues/58#issuecomment-375498107

theckman, 2020-10-17:
@bennuttall running into this issue on an RPi that I needed to run Ubuntu Server on. I'm trying to use the Python module to validate my own Sense driver I'm writing in another language, and unfortunately I'm getting all zeroes back for only the Python code.

Where can I find out more about what's in the `sense-hat` package to see if there's something there that will help fix this up for me?

beamerblvd, 2020-10-21:
@theckman, I'm encountering the same issue. I'm running Ubuntu Server, just like you, and all my sensor values are 0. This is what I did:

- Installed G++
- Cloned https://github.com/RPi-Distro/RTIMULib
- Created a virtualenv in Python for using Sense Hat
- Ran `python setup.py build` and `python setup.py install` in RTIMULib/Linux/python to compile and install the RTMI Python bindings
- Ran `pip install sense-hat`
- Opened a Python shell
- Imported and instantiated `SenseHat`
- I can successfully use `show_message` to display messages on the 8x8 screen on the Sense Hat, but all sensor methods (`get_temperature`, `get_humidity`, `get_pressure`, etc.) return 0.

Some guidance on how to get this to work / what to try next would be appreciated. Thanks!

theckman, 2020-10-21:
@beamerblvd Thank you for clarifying your steps; I probably did pretty much the same as you.

jwhendy, 2020-10-27:
RTIMULib (and forks) seem to have no issues for whatever reason, and that trend continues with the recently discovered RTIMULib2, though one does. I just posted an inquiry there to see if we can find out why none of the sensors are working. If you look at [my post there](https://github.com/jeff-loughlin/RTIMULib2/issues/2), you can see an attempt to manually walk through the sense hat initialization of pressure/humidity sensors [here](https://github.com/astro-pi/python-sense-hat/blob/master/sense_hat/sense_hat.py#L77).

I can't get them to initialize. I have the same experience as @beamerblvd : accel reads, lights can be changed, but none of the other sensors are outputting. Depending on how I do it, I get 0 or the `OSError`.

fjctp, 2020-11-01:
I am encountering the same issue. I am running **Ubuntu Server 20.04.1**. It is a fresh install.

### Issues

1. I got this error when I run `get_temperature()`, `get_humidity()`, or `get_pressure()`.

```
OSError: XXXX Init Failed
```

2. I got zeros reading when I run `get_accelerometer_raw()`, `get_gyroscope()`, and `get_compass()`

### Debug steps

1. Check if i2c is working
```
ubuntu@ubuntu:~$ ls -al /dev/i2c-1
crw-rw---- 1 root i2c 89, 1 Apr  1  2020 /dev/i2c-1
```

```
ubuntu@ubuntu:~$ cat /boot/firmware/syscfg.txt
# This file is intended to be modified by the pibootctl utility. User
# configuration changes should be placed in "usercfg.txt". Please refer to the
# README file for a description of the various configuration files on the boot
# partition.

enable_uart=1
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on
cmdline=cmdline.txt
```

```
ubuntu@ubuntu:~$ i2cdetect -F 1
Functionalities implemented by /dev/i2c-1:
I2C                              yes
SMBus Quick Command              yes
SMBus Send Byte                  yes
SMBus Receive Byte               yes
SMBus Write Byte                 yes
SMBus Read Byte                  yes
SMBus Write Word                 yes
SMBus Read Word                  yes
SMBus Process Call               yes
SMBus Block Write                yes
SMBus Block Read                 no
SMBus Block Process Call         no
SMBus PEC                        yes
I2C Block Write                  yes
I2C Block Read                   yes
```

```
ubuntu@ubuntu:~$ i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- UU -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- UU -- -- 5f
60: -- -- -- -- -- -- -- -- -- -- 6a -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

2. add user to i2c group
```
sudo addgroup ubuntu input
sudo addgroup ubuntu i2c
```

3. Tried to access sensor data with `sudo python3 test_script.py`. Still had the same issues

4. Build `RTIMULib` from source
```
git clone https://github.com/RPi-Distro/RTIMULib
cd RTIMULib/Linux/python
python setup.py build
python setup.py install
```

jwhendy, 2020-11-01:
@fjctp I think we're at the mercy of someone with `RTIMULib` knowledge assisting. Unfortunately, of the many forks, I only found one with an issues tab. I [posted there](https://github.com/jeff-loughlin/RTIMULib2/issues/2). If you look at where the sensors are initialized in the code here, and manually reproduce these steps with `RTIMU` directly, you'll see they aren't being initialized. So, it's not exactly a sense-hat thing, per se, it's that this key library is not functioning.

bsimmo, 2020-11-01:
>    1. I got zeros reading when I run get_accelerometer_raw(),
>    1. add user to i2c group

Does this help, notice the blacklist (ubuntu user)
https://github.com/J-Pai/EnvTrackerNode

fjctp, 2020-11-01:
@bsimmo Thank you very much! The blacklist solved the problem.

fjctp, 2020-11-01:
> @fjctp I think we're at the mercy of someone with `RTIMULib` knowledge assisting. Unfortunately, of the many forks, I only found one with an issues tab. I [posted [...]

Correct. I complied some examples in `RTIMULib` and found that it couldn't find the sensors in i2c. I think it is related to the driver. Look at the link posted by @bsimmo. That fixed the problem.

jwhendy, 2021-01-11:
@bsimmo and @fjctp I tried the blacklisting on arch and it did not work for me. It does work as root, however. This is a little bit annoying as I have to install python libs to the system folders, and I generally use `--user` for built packages.

This was pretty cool to see after years of not being able to do so:
```
# python sense_hat_demo.py 
Temperature: 20.3 °C - 68.5 °F
Humidity: 40.1 %rH
CPU Temperature: 39.2 °C
p: 2.54382626296944, r: 295.6990145496125, y: 179.13083611631552
North: 179.13083611631552
p: 2.589574229699951, r: 295.71873330574635, y: 179.11773581348746
p: 2.5504402339635397, r: 295.95182034136224, y: 179.12165634207102
```

Now, that said... other than this, what else can I check with respect to why this would only work as root?
```
$ cat /etc/udev/rules.d/99-i2c.rules 
KERNEL=="i2c-[0-7]",MODE="0666"
```

The EnvTrackerNode readme states "Application requires sudo" as a possible problem, but doesn't state how to fix it. I'm posting an issue there, but would appreciate any other suggestions here. I'm actually not sure if it was the blacklisting or root use that actually fixed this for me in the end.

J-Pai, 2021-01-11:
Just wanted to leave my $0.02 here. The reason for the blacklist is b/c the RPi build of Ubuntu Server 20.04.1 comes with the Industrial I/O modules preloaded.
Without the blacklist, you get a initialization errors on some of the sensors. I think this has something to do with the fact that the Industrial I/O modules grab on to the i2c sensors on the Sense HAT and refuse to let them go or allow them to be read correctly.

To resolve this issue the blacklist disables those drives and allows the Sense HAT drivers (rpisense_fb) to actually pick up the sensors.

I figured this out via lsmod. Since @jwhendy is using Arch (not Ubuntu), this may not actually impact his use case since I'm not sure if arch has those modules pre-installed.

Example lsmod output:

```
ubuntu@ubuntupi:~$ lsmod | grep st_
st_magn_spi            16384  0
st_pressure_spi        16384  0
st_sensors_spi         16384  2 st_pressure_spi,st_magn_spi
st_magn_i2c            16384  0
st_pressure_i2c        16384  0
st_pressure            20480  2 st_pressure_i2c,st_pressure_spi
st_magn                20480  2 st_magn_i2c,st_magn_spi
st_sensors_i2c         16384  2 st_pressure_i2c,st_magn_i2c
st_sensors             28672  6 st_pressure,st_pressure_i2c,st_magn_i2c,st_pressure_spi,st_magn,st_magn_spi
industrialio_triggered_buffer    16384  2 st_pressure,st_magn
industrialio           94208  9 st_pressure,industrialio_triggered_buffer,st_sensors,st_pressure_i2c,kfifo_buf,st_magn_i2c,st_pressure_spi,st_magn,st_magn_spi
```

If you notice the module names noted in the output above, they are associated with the problematic sensors that fail to be initialized. With the blacklist, these modules will not be loaded and the sensehat_fb (and associated modules) will be able to work as expected.

Not sure why the ARM/RPi version of Ubuntu has this module pre-installed, but at least we have the option of disabling them.

jwhendy, 2021-01-11:
I'll try un-blacklisting those on arch to see if I note any effect. With the insight on root and i2c permissions, I haven't A/B tested to check the black list. I can confirm from memory that `insutrialio*` were loaded, but I don't think I checked for `st_*` before blacklisting and rebooting. On arch, there were also `hts221*` modules and `kfifo_buf` that were loading `industrialio` and had to be blacklisted.

jwhendy, 2021-01-11:
K. Confirming blacklist matters on arch as well.
```
$ lsmod |grep st_
st_pressure_spi        16384  0
st_magn_spi            16384  0
st_sensors_spi         16384  2 st_pressure_spi,st_magn_spi
regmap_spi             16384  2 st_sensors_spi,hts221_spi
st_pressure_i2c        16384  0
st_magn_i2c            16384  0
st_pressure            16384  2 st_pressure_spi,st_pressure_i2c
st_magn                16384  2 st_magn_i2c,st_magn_spi
st_sensors_i2c         16384  2 st_magn_i2c,st_pressure_i2c
st_sensors             24576  6 st_magn_i2c,st_pressure_spi,st_pressure,st_magn_spi,st_pressure_i2c,st_magn
industrialio_triggered_buffer    16384  3 hts221,st_pressure,st_magn
industrialio           73728  10 hts221,st_magn_i2c,st_sensors,st_pressure_spi,st_pressure,st_magn_spi,st_pressure_i2c,industrialio_triggered_buffer,st_magn,kfifo_buf
```

Woohoo, I think this can actually be closed! tl;dr for the future:

- blacklist modules as described at [EnvTrackerNode](https://github.com/J-Pai/EnvTrackerNode)
- in addition to checking `/etc/udev/rules.d/*` for `/dev/i2c-1` access, also consider the `i2c` group method referenced [here](https://github.com/J-Pai/EnvTrackerNode/issues/1#issuecomment-758161167). A symptom of this will be the ability to run the `EnvTrackerNode/scripts/sense_hat_demo.py` only as root/with sudo
- tbd: whether or not building RTIMULib and python-sense-hat are required vs. installing packages, which I'll leave to someone else to test

closed 2021-01-11 by jwhendy