astro-pi/python-sense-hat #81: Accented chracters in sense.show_message()

g-vidal, 2017-10-12:
Hi,
I am trying to write some french text on the sense-hat led matrix, it works perfectly for unaccented ASCII characters but I cannot write french accented characters, while the print() commands works perfectly. 

```
# -*- coding: utf-8 -*-

from sense_hat import SenseHat
sense = SenseHat()
sense.set_rotation(180)
sense.low_light = True
sense.show_message('Fête de la science', scroll_speed=0.05, text_colour=[0, 100, 128])
```

Do I miss something or some work is needed?
thanks

bsimmo, 2017-10-16:
I believe the characters come from a picture file (.PNG). If it's not in there, then you cannot display it.
 (See some other open problems).

Probably a problem now the Astro-Pi project is open to more than just the UK.

byackley, 2018-01-17:
I'm working on a solution to this. I'll have a pull request up in the next day or two.