astro-pi/python-sense-hat #84: Unable to Detect Sense Hat: OSError: Cannot detect RPi-Sense FB device

plutosrings, 2018-03-18:
Hello,
Thank you in advance for any help offered. I know others have seen this issue unfortunately none of their solutions have worked for me =/

When trying to run the python code:
   from sense_hat import SenseHat
   a=SenseHat()

I'm getting: 
Traceback (most recent call last):
 ....
OSError: Cannot detect RPi-Sense FB device

This is with the raspberrypi 3-model-b using Arch Linux on ARM

$uname -a 
Linux cm02-mon 4.15.10-1-ARCH #1 SMP Fri Mar 16 18:10:31 MDT 2018 aarch64 GNU/Linux

I have installed the packages from the AUR after re-configuring both PKGBUILDS to support aarch64:
     python-rtimulib-7.2.1-3-aarch64.pkg.tar.xz
     python-sense-hat-2.2.0-3-aarch64.pkg.tar.xz
Both built & installed without issue

I have tried the typical /boot/config.txt additions:
dtparam=i2c_arm=on

have also tried:
dtparam=i2c_arm=on
dtoverlay=rpi-sense

To no avail, the hat is secure to the top of the 40 pin connector. I have tested this with 2 separate Hats that were purchased together; both exhibit this issue. 

dmesg output:
https://pastebin.com/tncx6TRH

dtc output:
https://pastebin.com/5jzsj74j

Again, any help is appreciated. Thank you so much!

bsimmo, 2018-03-18:
Personally I would test the HATs work using Raspbian, since that is the
supported OS and has all the gubbins preinstalled.
Get the SenseHAT working there and then move to another OS.
It should just be boot to the desktop and run Python3.

If it's the OS then post in the forums.

plutosrings, 2018-03-18:
Hello Bsimmo,
I was able to the hats working with Rasbian using the rasbian-stretch img:

dmesg | grep sense
[    3.899367] rpi-sense 1-0046: Raspberry Pi Sense HAT firmware version 0

Python code works as expected, got messages across the LED matrix. 

I'll follow up here: https://archlinuxarm.org/forum/index.php regarding support for Arch on ARM support, unless there is somewhere else you'd suggest looking for help.

plutosrings, 2018-03-18:
I have posted to archlinuxarm forums here: 
https://archlinuxarm.org/forum/viewtopic.php?f=65&t=12646

Hopefully we can figure whats missing or going on, thanks again!

plutosrings, 2018-03-18:
Was able to get the Hats working! Had to use armv7 vice the latest armv8/aarch64:
found here: https://archlinuxarm.org/platforms/armv7/broadcom/raspberry-pi-2

Even though its intended for the older Pi 2, it appears to work well on the Pi 3.

I did have to run the raspi-config tool (in the AUR) to enable the i2c interface. Seems pretty stable in this version.

Writing to the matrix on both Hats now, and quite happy. Feel free to close this issue here, I'll follow up with the folks over in the Archlinux ARM thread regarding  sense-hat support there. 

Cheers!

closed 2018-03-18 by XECDesign