astro-pi/python-sense-hat #85: spelt colour wrong

Aoxo, 2018-04-01:
My American brain is deeply troubled by the way the British spell certain words.  Normally I can let it go but I know I'm going to get really confused with stuff like this:

`sense.show_message("Stella is a bad cat.", text_colour=textColor, back_colour=backColor)`

Is it possible to use localiZation to translate any common spelling differences in the api?  Add aliases or something?  Or will I just have to adapt to the queens english?

bsimmo, 2018-04-01:
Adapt, we have to do it the other way all the time, so annoying. It's nice
to have it the English way for a change.
;-)
Ok so it's not up to me.

AlmightyOatmeal, 2018-04-07:
Aoxo, 

Aside from your rudeness and ignorance, you can make the changes yourself. Simply clone the repo, copy the `sense_hat` directory to your project’s directiry, and make the edits you deem necessary. Because the module is in your projects base directory, your import of `sense_hat` will use your local copy before checking the other library paths.

Of course you could submit a pull request with your changes too so that other people who are offended by the British can benefit as well.

While you’re making the change from `colour` to `colour`, would you mind creating an alias to `tsvet`? Russian is my native language (even though I have lived in the states for most of my life) and your American ignorance offends me so why not add an alias to satisfy my language requirements? Also, can you add `farbe` because my adoptive grandfather was German? And `farge` because my adoptive grandmother was Norwegian?

It’s only fair, right?

The easiest solution is to grow up but that is unfortunately difficult for some people to grasp.

Aoxo, 2018-04-09:
-AlmightyOatmeal
I asked a question.  I didn't insult anyone or any nationality and if I did I surely didn't mean to.  Ignorant?  Yes, well it WAS a question wasn't it?  Am I expected to know the answer to my own question?

You just went on a long angry rant insulting me and my country for no apparent reason, yet you call me rude?

I'm sure the issues section for the Sense hat API isn't the place to vent your hostility.  Even if you did somehow manage to weave an answer to my question into your string of insults.

AlmightyOatmeal, 2018-04-09:
Aoxo,

You have a solution yet you still complain. 

You don’t feel your question was insulting when it was; you have no right to tell someone else how your words come across; that shows ignorance. You should choose your words carefully next time.

My words were more than fair — if you don’t like `colour` and want it changed to `color` then you should consider other nationalities.

The whole premise to this issue is absolutely insulting and the fact you can’t see that only affirms your ignorance.

Aoxo, 2018-04-10:
You are not the authority on what my words mean, I AM.  Your perception of what they mean to you is yours, but that's it.  It is not unreasonable for me to ask if there is a way to work around my language issue.  I didn't ask all of the United Kingdom to stop using the word Colour.  I didn't ask the developers to add or change features.

I'll just shoot to the bottom line here, because any more discussion with you would just be feeding a troll.  You're a stereotypical condescending elitist nerd.  You can take whatever offense you took and shove it.  I don't really care if you were offended or not because I think any normal person reading my post was able to process my question without taking offense to the notion of translation.  How dare you criticize me for my nationality and my language?

I'm sorry that your existence is so sad that you must scour the Internet for things to rage about and demean people who know less than you.  You really need to get a life.

Aoxo, 2018-04-10:
To everyone else:
I'm sorry that I've been so far unable to resist a troll, an that the Sense Hat issues forum has turned into a nerd fight.  If you're reading this then I'm sure that drama isn't what you clicked for.  Hopefully no one is reading, and if you are...I pity you.

I got a Sense Hat to learn more about code, not because I already know everything.  I'm sure this is true for a number of the people who use the product.  It's pretty sad that some people feel a need to personally attack people who ask questions, if even dumb questions.  This kind of thing happens alot.  It does very little to encourage people to challenge themselves to learn more about something technical when asking a simple question somehow becomes an invitation for personal insults from a socially inept know-it-all.

closed 2018-04-10 by Aoxo

bsimmo, 2018-04-10:
Being someone who replied and hence gets to read these.
I am from the UK, I am English.
Given we say the same about the way you (Generalised Americans) mangle the
language with you lack of U's, Z instead of S's and other simplifications,
I didn't take offence, but read it in good humour.

It is a problem as we normally have to cope the other way around, teaching
kids to spell colour as colour and then have to type color.

It's a good suggestion.

P.S. there is no reason other languages could be added, but it takes
someone to do the 'fix' and create a pull request.

One thing though this is used as standard code on the ISS and used though
schools in then UK, EU with respect to to the ESA projects.
So I don't think changes will be added lightly, mainly as changing it on
the ISS maybe a bit troublesome?  At least not while competitions are
running.

But the RPF maintainer(s) may respond at some point.

I know there is another topic on adding/changing other language characters
to the text image files for the scrolling text.

Have fun and no offence taken or hopefully given :-)