astro-pi/python-sense-hat #87: sense HAT & baud rate settings?

radiogithub, 2018-10-29:
Hi there

How can I get the sense HAT working with increased I2C baud rate?

I started with an inverted pendulum project were I have a RPI 3 B+, a RPI sense HAT and an Adafruit Motor HAT. I stacked the RPI sense HAT on the GPIOs of the RPI 3 B+ and the Adafruit Motor HAT on the GPIOs of the sense HAT (I use an extra tall stacking header from Adafruit). At I2C default baud rate my stepper motors jerk quite significantly. Thus, I set the I2C baud rate to 1000000 to improve the situation. When I increase the I2C baud rate the sense HAT is not detected and I get the error message below. Actually when I power up the RPI the LEDs of the sense HAT stay on. The Adafruit motor HAT works much better than with the default I2C baud rate (even though not perfect yet).
OSError: Cannot detect RPi-Sense FB device

I set the baud rate with following line in the /boot/config.txt file:
dtparam=i2c_arm_baudrate=1000000

Stacking the sense HAT and the Adafruit Motor HAT works if the default I2C baud rate is set (by uncommenting the above line) and with line below in the boot/config.txt file. So its not a hardware issue. However. I have read in this blog to not use this line.
dtoverlay=rpi-sense

Further details:

Adafruit Motor HAT
https://www.adafruit.com/product/2348

Stepper Motors
https://www.u-buy.ch/en/catalog/product/view/id/1514538/s/stepping-motor-nema-17-stepping-motor-26ncm-36-8oz-in-12v-0-4a-3d-printer-cnc/

Output of command
sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- 1c -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- 46 -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- 5f 
60: 60 -- -- -- -- -- -- -- -- -- 6a -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --  

(I guess 60 & 70 are the motor HAT address. The others are for the sense HAT.)

Output of command
uname -a
Linux raspberrypi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l GNU/Linux

Output of command
vcgencmd version
Aug 16 2018 17:30:30 
Copyright (c) 2012 Broadcom
version 31e0613622dc2f2463bf3dd74e6c897d91201a4d (clean) (release)

Output of command
sudo dtc -I fs /proc/device-tree
<stdout>: Warning (unit_address_vs_reg): Node /soc has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/vchiq has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/gpiomem has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /soc/axiperf has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /memory has a reg or ranges property, but no unit name
<stdout>: Warning (unit_address_vs_reg): Node /axi/vc_mem has a reg or ranges property, but no unit name
<stdout>: Warning (avoid_default_addr_size): Relying on default #address-cells value for /axi/vc_mem
<stdout>: Warning (avoid_default_addr_size): Relying on default #size-cells value for /axi/vc_mem
/dts-v1/;

/ {
compatible = "raspberrypi,3-model-b-plus", "brcm,bcm2837";
serial-number = "0000000075c97e61";
model = "Raspberry Pi 3 Model B Plus Rev 1.3";
memreserve = <0x3b400000 0x4c00000>;
interrupt-parent = <0x1>;
#address-cells = <0x1>;
#size-cells = <0x1>;

clocks {
 compatible = "simple-bus";
 #address-cells = <0x1>;
 #size-cells = <0x0>;
[... 1541 lines cut ...]
[    6.096706] brcmfmac: power management disabled
[    6.372591] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.387699] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    7.786309] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   12.348426] Bluetooth: Core ver 2.22
[   12.348520] NET: Registered protocol family 31
[   12.348526] Bluetooth: HCI device and connection manager initialized
[   12.349493] Bluetooth: HCI socket layer initialized
[   12.349510] Bluetooth: L2CAP socket layer initialized
[   12.349548] Bluetooth: SCO socket layer initialized
[   12.367320] Bluetooth: HCI UART driver ver 2.3
[   12.367340] Bluetooth: HCI UART protocol H4 registered
[   12.367346] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   12.367581] Bluetooth: HCI UART protocol Broadcom registered
[   12.693805] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.693814] Bluetooth: BNEP filters: protocol multicast
[   12.693826] Bluetooth: BNEP socket layer initialized
[   12.747394] Bluetooth: RFCOMM TTY layer initialized
[   12.747442] Bluetooth: RFCOMM socket layer initialized
[   12.747461] Bluetooth: RFCOMM ver 1.11
[   13.561705] fuse init (API version 7.26)

Output of command
ls /dev/fb*
/dev/fb0

XECDesign, 2018-10-30:
If I remember correctly, the clock is intentionally slowed down to give the firmware on the HAT enough time to avoid clock stretching and avoid hitting the clock stretching bug.

mme384, 2018-10-30:
Thank you for your answer. I.e., I can not increase the I2C baud rate. Any other suggestion how to improve the performance of the stepper motor HAT?

Thanks.