astro-pi/python-sense-hat #88: RTIMULib copy should be optional

teuteuguy, 2018-12-07:
Copying the RTIMULib to the home folder requires access to the home folder, which in some cases, ie. running as root can become complex.

This should be a setting on load of the lib, to choose whether or not to copy the RTIMULib file or not.

Event, be able to provide path to file directly.

XECDesign, 2018-12-07:
In general, you should be guaranteed access to your home directory, if nowhere else. But yes, flexibility wouldn't hurt.

J-Hurtta, 2020-01-30:
Indeed. I'm trying to use python-sense-hat on on a limited access case (Home Assistant on Hass.io).

As the RTIMULib is not listed as a dependency, I can't install it on my device and thus cannot use my sense hat.

rzr, 2020-05-06:
or at least  requierement should be mentionned somewhere