astro-pi/python-sense-hat #96: Buster won't boot headless with the Sense Hat attached

svanya, 2019-07-02:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=243949
--------------------------------------------------------
I have a raspberry Pi 3B+ with a sense hat attached. I installed buster on a new card (enabled ssh) and it boots up with no problems if it is attached to a display. When I try to boot headlessly the Hat just lights up the rainbow and stays that way. The Pi does not seem to boot up. With the Hat off it boots headlessly and is accessible via ssh.

I have repeated this with a second sd card and on the Pi 3B+ and 2B. I have tried disabling the splash screen and graphical boot to no avail.
------------------------------------------------------
I have RPI 4 and have the same problem with Sense Hat. I spent hours on it.
Why there is no testing before launch?
------------------------------------------------------
Same thing here - I'm using a RPi 3B+ with a SenseHat attached. It won't boot Buster in headless mode, it'll only boot with a display connected. Also tried disabling splash screen and using the legacy OpenGL driver, to no avail - the Sense LED Matrix stays lit up, just like it would if I would not have any SD card inserted.
------------------------------------------------------
Same thing here with Sense Hat + RPi 3B+
------------------------------------------------------

XECDesign, 2019-07-02:
Of course there is testing before launch. It's just that every single combination of hardware and configuration is not possible to test.

A workaround is to set a monitor resolution in raspi-config.

PBernalPolo, 2019-07-24:
I had the same problem with buster RPi 4B. As @XECDesign states, doing:

**$raspi-config**
**7 Advanced Options**
**A5 Resolution**
**DMT Mode 4 640x480 60Hz 4:3**

solved the problem. Tested with:
- RPi4B
- 2019-07-10-raspbian-buster
- 2019-07-10-raspbian-buster-lite