astro-pi/python-sense-hat #97: SenseHAT all LEDs On with Hardware Not Found

avishek-rc1, 2019-07-14:
**Description:** 
This was working perfectly earlier. All of a sudden, when Pi boots up all LEDs are on and nothing happens. Read a lot of forums but no specific answer. Please help.

**Error:** 
Traceback (most recent call last):
  File "/home/pi/Projects/SenseHat/sensehat_sensing.py", line 2, in <module>
    sense = SenseHat()
  File "/usr/lib/python3/dist-packages/sense_hat/sense_hat.py", line 39, in __init__
    raise OSError('Cannot detect %s device' % self.SENSE_HAT_FB_NAME)
OSError: Cannot detect RPi-Sense FB device

**Attached: vcdbg log msg and dmesg**
[vcdbg.txt](https://github.com/RPi-Distro/python-sense-hat/files/3390004/vcdbg.txt)
[dmesg.txt](https://github.com/RPi-Distro/python-sense-hat/files/3390005/dmesg.txt)

XECDesign, 2019-07-15:
All on meaning all white or a rainbow pattern?

It sounds like a hardware issue. The Pi is unable to communicate with the Sense HAT, so there's not much to go on.

avishek-rc1, 2019-07-15:
Its a rainbow pattern. Does this mean the HAT is destroyed? Is there a workaround? Changing the breakout pins etc.?

XECDesign, 2019-07-15:
I don't know if it's destroyed or if there's just dirt in the connector. Without knowing what's actually wrong, there's not much that can be done. If it's still within the warranty period, you could try to RMA it.

bsimmo, 2019-07-15:
Try removing and reconnecting it a few times. You could also take the black
cover off the pin protection and check it contacts well.

If that fails then force detection with
add dtoverlay=rpi-sense
 in sudo nano /boot/config.txt
reboot

bsimmo, 2019-07-15:
Ah a quick look through you dmesg, the sense fails to be detected when it tries, are you powering via OTG-USB?
Anyway, undervoltage etc at the end.

Same reason I had, not enough power during boot time.
Sort your power out. 
See if it works.

avishek-rc1, 2019-07-15:
Thank You Ben. I am powering using the standard 5v/2.5a power outlet only. Also, the dt-overlay seems to be added in my config. Attaching a copy of it. 
[config.txt](https://github.com/RPi-Distro/python-sense-hat/files/3392957/config.txt)

bsimmo, 2019-07-15:
It'll still be power, mainly because it is warming you.
The 3B+ seems very sensitive to anything during boot and the SenseHAT just
stretches it too much, perhaps the LEDs just draw that but too much and
make the Pi PSU drop out with a spike.
But I can confirm you are not the only one, as it happened to me.

You could try the newer Buster setup and see if that has a better boot up
sequence. That is if you have no other PSUs to try.
(Yes iirc mine was an official one too).

avishek-rc1, 2019-07-15:
Thank You Ben, I will try with a new PSU first and if it still fails, would try with Buster. Might take a day or two to respond on this thread again.

bsimmo, 2019-07-15:
For reference

```
[    4.459523] rpi-sense 1-0046: Read from reg 240 failed
[    4.459578] rpi-sense: probe of 1-0046 failed with error -22
[    4.884412] uart-pl011 3f201000.serial: no DMA platform data
[    5.119991] random: crng init done
[    5.120011] random: 7 urandom warning(s) missed due to ratelimiting
[    5.397594] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    5.397614] brcmfmac: power management disabled
[    5.592001] Under-voltage detected! (0x00050000)
```
And then at the very end of dmesg, some in and out of voltage messages.