astro-pi/python-sense-hat #102: Numpy error running sense-hat module on 3.9.0

moritzsalla, 2020-10-22:
Is it possible to run sense-hat on 3.9.0? If not, what is the highest python version it can run on?

```
$ python main.py  

python(53983,0x10c182dc0) malloc: can't allocate region
:*** mach_vm_map(size=18446744072444301312, flags: 100) failed (error code=3)
python(53983,0x10c182dc0) malloc: *** set a breakpoint in malloc_error_break to debug
init_dgelsd failed init
Traceback (most recent call last):
  File "/Users/moritzsalla/Documents/GitHub/raspi-wekinator-client/app/main.py", line 6, in <module>
    from sense_hat import SenseHat
  File "/Users/moritzsalla/.pyenv/versions/3.9.0/lib/python3.9/site-packages/sense_hat/__init__.py", line 2, in <module>
    from .sense_hat import SenseHat, SenseHat as AstroPi
  File "/Users/moritzsalla/.pyenv/versions/3.9.0/lib/python3.9/site-packages/sense_hat/sense_hat.py", line 7, in <module>
    import numpy as np
  File "/Users/moritzsalla/.pyenv/versions/3.9.0/lib/python3.9/site-packages/numpy/__init__.py", line 286, in <module>
    raise RuntimeError(msg)
RuntimeError: Polyfit sanity test emitted a warning, most likely due to using a buggy Accelerate backend. If you compiled yourself, see site.cfg.example for information. Otherwise report this to the vendor that provided NumPy.
RankWarning: Polyfit may be poorly conditioned
```

moritzsalla, 2020-10-23:
Ok, I got it to work on Thonny running Python 3.7.3 (/usr/bin/python3). However, running my script in terminal using pyenv 3.9 (/home/pi/.pyenv/shims/python) gives me this error:

```
ModuleNotFoundError: No module named 'sense_hat'
```

I'm installing 3.7.3 on pyenv to see if that will run it.

moritzsalla, 2020-10-23:
After finding this page: https://www.raspberrypi.org/forums/viewtopic.php?t=154230, I followed all steps, but I'm still stuck with either the "missing module" or "Importing the numpy C-extensions failed" error.

I tried:
```
sudo apt-get remove python-sense-hat python3-sense-hat
sudo pip install sense_hat
sudo pip3 install sense_hat
```

and: 
```
sudo apt-get update
sudo apt-get install python-sense-hat python3-sense-hat -y
```

```
```

wingkitlee0, 2020-12-26:
Your original numpy error said `libf77blas.so.3` is missing. It can be installed by running:
```
 sudo apt install libatlas-base-dev
```

closed 2020-12-30 by moritzsalla