astro-pi/python-sense-hat #104: Doing Maths with Pixel Values

rowena-s, 2020-12-12:
I have a design for the LED array where I want to slowly pulse the lights by increasing and decreasing intensity of specific pixels. However, when I get the values with get_pixel() it seems to always be less than what it should actually be.
`
sense.set_pixel(4,0,[255,255,0])
r,g,b = sense.get_pixel(4,0)
print(r,g,b) # shows 248 252 0 when it should be 255 255 0

r1 = r-20
g1 = g-20
print(r1,g1,b) # shows 228, 232, 0 which is correct from the actual values

sense.set_pixel(4,0,[r1,g1,b])
print(sense.get_pixel(4,0)) #  shows me 224, 232, 0
`

Is there a way to achieve what I want given I can't rely on getting back the correct pixel values?

bsimmo, 2020-12-12:
What you are seeing is correct.

Use a variable(s) in your program to store the values and always use that
rather than using the led itself.

rowena-s, 2020-12-12:
ah, thanks!

closed 2020-12-12 by rowena-s

bsimmo, 2020-12-12:
Bit more time now, but there is a short explanation of what you were seeing
on the API Reference - Sense HAT (pythonhosted.org)
<https://pythonhosted.org/sense-hat/api/#led-matrix> page, look down to
getpixels()

rowena-s, 2020-12-12:
ohhh, I looked at getpixel() - singular, and missed the note at the bottom that says to check the note under the plural version.