astro-pi/python-sense-hat #107: .txt

ghunter717, 2021-03-08:


rzr, 2021-03-23:
Please edit or close this ticket , btw is this project still maintained if not it could be xfered to @abandonware org

https://rzr.github.io/rzr-presentations/docs/abandonware

closed 2021-03-23 by ghunter717

bsimmo, 2021-03-23:
> Please edit or close this ticket , btw is this project still maintained if not it could be xfered to @abandonware org
> 
> https://rzr.github.io/rzr-presentations/docs/abandonware

It can't change much as it is used by the ESA, RPF etc for various AstroPi competitions.
I believe updating the two Pi's with SenseHATs on the ISS isn't so easy and school and groups use a duplicate of that setup.

rzr, 2021-03-23:
ok thx for feedback, I have opened a couple of PR though:
https://github.com/astro-pi/python-sense-hat/pulls/rzr