astro-pi/python-sense-hat #121: Unable to initialize old Sense HAT after update to 2.3.1

dbenesj, 2022-07-08:
After the update to version 2.3.1 I'm not any more able initialize the library with the old/original/v1 Sense HAT.
It complains that TCS34725 color sensor is not available.

```
$ uname -a
Linux malina4 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```
```
$ cat /etc/*release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
```
$ python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44)
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from sense_hat import SenseHat
>>> sense = SenseHat()
WARNING:root:Failed to initialise TCS34725 colour sensor. (sensor not present)
```

Is there some way how to tell the library to not try to initiate the color sensor?

G3zz, 2022-07-08:
Hi,
This is just a warning, and not an error - you can still use the library as it was in previous versions
Geraint

dbenesj, 2022-07-08:
Solved by changing the logging level
```
...
import logging
from sense_hat import SenseHat
logging.getLogger().setLevel(logging.ERROR)
sense = SenseHat()
logging.getLogger().setLevel(logging.WARNING)
...
```